OctoPrint/OctoPrint #1744: Octoprint closes the connection to the printer on "MNTEMP trggered"

rewolff, 2017-02-01:
I think that the prnter I'm bulding has a loose connecton to the thermistor in the hotend. This now results in "mintemp triggered" errors. and octoprint dsconnectnig. On reconnect the same happens, so it is difficult to debug the problem because  I cant use ocotprnt to connect/control/diagnose the prnter. 

#### What were you doing?

debugging a new printer. 

#### What did you expect to happen?
ocotrpint should connect and display the 'mntenp trgered as an error"

#### What happened instead?
octoprint dsconnnects and dsplays mntemp triggered.
#### Branch & Commit or Version of OctoPrint
Version 1.3.0.post1+g1014712 (master branch)

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version
reprap beagle copy wth a few modficatohns. we forgot to saw off the rods at the required length, so we ended up wth 570 Z-buildvolume 
Marln compiled from source.

#### Browser and Version of Browser, Operating System running Browser
chromium not relevant. 

#### Link to octoprint.log
i cant find a logfile
#### Link to contents of terminal tab or serial.log
onnection closed, closing down monitor
Changing monitoring state from 'Detecting baudrate' to 'Offline'
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x7f6850091090, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Send: N0 M110 N0*125
Connection closed, closing down monitor
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 230400
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 115200
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 57600
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 38400
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 19200
Send: N0 M110 N0*125
Baudrate test retry: 4
Send: N0 M110 N0*125
Baudrate test retry: 3
Send: N0 M110 N0*125
Baudrate test retry: 2
Send: N0 M110 N0*125
Baudrate test retry: 1
Send: N0 M110 N0*125
Baudrate test retry: 0
Send: N0 M110 N0*125
Trying baudrate: 9600
Send: N0 M110 N0*125

foosel, 2017-02-01:
Uncheck "Settings > Serial > Not only cancel ongoing prints but also disconnect on unhandled errors from the firmware". That should keep it from disconnecting on your mintemp error.

Also if you know the baud rate of your printer, select that on connection to ensure it going faster and not doing the lengthy auto detection.

Solved? Then please close the issue :)

rewolff, 2017-02-01:
OK. I can now connect and see that the bed is 14.9 (thermostat of my room s set for 15.5 so close enough) and that t thnks the hotend is 30 degrees (whle in fact t too is at room temp). (sorry, my keyboard has a flaky key...)

foosel, 2017-02-01:
You'll need to figure out why the firmware is reading the hotend temperature wrongly. This however is not something OctoPrint can solve, it relies on receiving correct data from the firmware :)

So can this be closed?

rewolff, 2017-02-01:
No, but I don't see why octoprint should refuse to stay connected with a printer that detects an error. Of course printing should not be attempted. 
This time, it seems octoprint stayed connected for a while and then got something "killed" and then no more communication. It seems I have the same gripe with marlin.... why stop communicating when there is an error?

MiggyMan, 2017-02-01:
MINTEMP is a shutdown error, octoprint isn't disconnecting anything, the printer is.

foosel, 2017-02-01:
As explained in my first comment, you can instruct OctoPrint to not disconnect *by itself* on an error by unchecking the mentioned setting. This setting is checked by default because the errors usually sent by firmware also are errors that will have the firmware go into kill state, effectively disconnecting and shutting down everything from the firmware side *until manually reset*. There's nothing OctoPrint can still do in that mode, so cleaning up the now dead connection is usually the correct approach. Also in case of your mintemp error. There is no way to use your printer until you have fixed that. Even printing from SD without OctoPrint in the mix is out of question since your firmware will enter kill state.

The behaviour of the firmware here is also the safe choice. A mintemp error indicates something wrong with the temperature probe which could easily lead to things like thermal runaway. Shutting off everything and signaling error is the safe way of handling that. Ignoring it would be a good way to start fires.

I'm closing this now since everything is behaving as expected and there is an easy way to change behaviour in case of errors sent by the firmware that do not lead to kill state.

closed 2017-02-01 by foosel

rewolff, 2017-02-01:
Agreed.  
Still it is stupid that the firmware Marlin will refuse to help me debug a flaky connection. Somehow it worked partially for a few minutes today (reporting 30 deg instead of about 15) and so now I'm back to long iterations between trying to reconnect with the malfunctioning printer and prodding the wire, while a live update: "temp is now reported as XXX" would be much quicker. 

Have you seen the code in marlin for this?
```
   if (!killed) {
      Running = false;
      killed = true;
      kill(lcd_msg);
    }
    else
      disable_all_heaters(); // paranoia
```

It only disables the heaters if it is already in "killed" state. It relies on being called again to disable the heaters.  If killed state somehow manages to prevent this function from getting called again, it will forget to disable the heaters.... Sigh!

Blue-Marlin, 2017-02-02:
`disable_all_heaters();` is also called in `kill()`. `kill()` ends in a endless loop with disabled interrupts, so is not supposed to exit. Calling `disable_all_heaters()` here again is really just caused by paranoia.

shupratik, 2020-02-22:
Check if the thermistor has any breaks , or it is burnt.