OctoPrint/OctoPrint #1745: Unable to update OctoPrint from 1.2.17 to 1.3.1

Runolas, 2017-02-01:
#### What were you doing?

Trying to update OctoPrint 1.2.17 to 1.3.1

#### What did you expect to happen?

Be able to use the latest version of OctoPrint

#### What happened instead?

Unable to update since my OctoPrint server

#### Branch & Commit or Version of OctoPrint

Version 1.2.17 (master branch)

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

Mozilla Firefox 51.0.1 (64 bits) for Ubuntu 16.04.1 LTS

#### Link to octoprint.log

https://gist.github.com/Runolas/d04207742764c74586031036a5004aa9

#### Link to contents of terminal tab or serial.log

https://gist.github.com/Runolas/e6f4f80b39b45ca18dbfec11cde16ce5

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

[image]

I have read the FAQ.

I have read the "note for upgraders" and I updated the init script : https://github.com/foosel/OctoPrint/releases/tag/1.3.1

Moreover, I have read the issues #1651, and I correctly configured my folders : https://github.com/foosel/OctoPrint/issues/1651

[image]

[image]

foosel, 2017-02-01:
> I correctly configured my folders

No, you did not. Your checkout folder is configured as `/home/Jeremy/Octoprint` up there. Your log however indicates that you have OctoPrint checked out at `/home/Jeremy/OctoPrint`. Linux file systems are case sensitive, meaning that you need to make sure you use the exact path, including the correct case. Try changing `/home/Jeremy/Octoprint` to `/home/Jeremy/OctoPrint` in your checkout folder configuration and that should hopefully get you going. Please report back and close the issue if that is the case.

Runolas, 2017-02-01:
Indeed, my checkout folder is "Octo**P**rint", and I wrote "Octo**p**rint". That a simple mistake ! Then, I updated correctly. Thank you very much !

closed 2017-02-01 by Runolas