OctoPrint/OctoPrint #1746: Cura 2.x slice settings in gcode file are in new format

imrahil, 2017-02-01:
#### What were you doing?
Cura 2.x (specifically 2.3.1) slicing settings are saved in new format than 14.x/15.x. Instead of 1 line at the end of gcode file there is block of lines with prefix:
`;SETTING_3 `

Example:

> ;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = Ustawienia zwykl
;SETTING_3 e\\ndefinition = custom\\n\\n[metadata]\\nquality_type = normal\\ntyp
;SETTING_3 e = quality_changes\\n\\n[values]\\nadhesion_type = skirt\\ninfill_be
;SETTING_3 fore_walls = False\\nlayer_height = 0.3\\ntop_bottom_thickness = 0.9\
;SETTING_3 \nspeed_travel = 70\\ninfill_sparse_density = 60\\nmaterial_diameter 
;SETTING_3 = 2.9\\nwall_thickness = 1.2\\nretraction_speed = 48\\nspeed_print = 
;SETTING_3 40\\ntop_bottom_pattern = concentric\\nskirt_gap = 10\\nlayer_height_
;SETTING_3 0 = 0.2\\nmaterial_bed_temperature = 50\\ninfill_pattern = lines\\nsk
> ;SETTING_3 irt_line_count = 3\\nretraction_amount = 0.4\\n\\n"}

This breaks gcode interpreter and variable 'filament_diameter' is not available.
My "Print History Plugin" is less usable now :)

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint
1.3.1

#### Printer model & used firmware incl. version

#### Browser and Version of Browser, Operating System running Browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

imrahil, 2017-02-08:
@armicron wrote Cura settings parser - https://github.com/imrahil/OctoPrint-PrintHistory/pull/43

cp2004, 2021-05-18:
Filament diameter is set in the printer profiles that's the only place I'm seeing it. This issue does not seem relevant anymore. OctoPrint doesn't parse gcode comments because they change regularly and there is no standard anyway.

closed 2021-05-18 by cp2004