OctoPrint/OctoPrint #1749: [Request] Confirmation dialog when deleting timelapses

ardenpm, 2017-02-02:
The delete and download option for timelapses are very close to one another. It is easy to hit the delete option by mistake and as there is no confirmation dialog (where as there are confirmations for other serious actions). This results in the file being immediately deleted.

A confirmation dialog similar to others seen in Octoprint would be grerat. If it is felt this would be too annoying perhaps it could be optionally enabled.

aolshove, 2017-04-12:
Agreed.  A delete confirmation is NEVER too annoying to implement.  Some might say that it's a basic necessity for any software project and timelapse delete confirmation is definitely a necessity.

foosel, 2017-04-12:
Tell that to the people who screamed their lungs off after I merged a print cancel confirmation ;) Things aren't always as black and white as they might look at first :)

With regards to this feature request, this is a duplicate of #748 (and I actually thought I'd already marked it as such) which has been implemented on `maintenance` and `devel` and that implementation will be released with 1.3.3.

foosel, 2017-04-12:
Also refer to the discussion in the brainstorming ticket #1807

closed 2017-04-12 by foosel

aolshove, 2017-04-12:
Weelll... I think that a delete confirmation is always a necessity.  People will complain about "one more button to click" but risk of lost data should never be compromised.  I just tell my users, "tough toenails, deal with it", because it's less fallout than them begging/blaming me to retrieve their lost data later.  

Sorry for missing the duplicate issue.  I missed it in the search and had assumed that if this was open, it was valid.  Thank you for your time.