OctoPrint/OctoPrint #1751: API key enabled but reads n/a in settings (but is present in config.yaml)

ntoff, 2017-02-03:
#### What were you doing?

messing about as usual, updating my "send to printer" batch file, so I went looking for my API key

#### What did you expect to happen?
the api key to be listed in the settings page of octoprint

#### What happened instead?
it just says n/a

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (master branch) - manual install following the instructions in the wiki (not my  first manual install either but see notes at bottom)

#### Printer model & used firmware incl. version

mendel90 running repetier (not sure that's relevant, printing functions 100%)

#### Browser and Version of Browser, Operating System running Browser

google chrome Version 55.0.2883.87 m
 on windows 10 pro

#### Link to octoprint.log

https://gist.github.com/ntoff/02a4cd61f7b594771a81ebca59312662

There's a bunch of them all rotated out after a certain amount of time, this is the most recent one, would you like *all* of them zipped up somewhere?

#### Link to contents of terminal tab or serial.log

*no soup for you!* I doubt it's related to serial comms but the printer is currently printing something so all that's in the serial tab is a tonne of G1's (printing is 100% functional)

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s) showing the problem:

[image]

It exists in config.yaml
[image]

Extras: I don't know if it's related in any way but I got the update notification so let octoprint update itself and it refused to start back up again. Not really sure whether it was just the reboot that fixed it or that I went through most of the install steps again. I could start it manually just fine without errors but the service refused to run. It seems happy now after a reboot but I can't be sure what exactly fixed it, however the API key just says n/a in the settings box but apparently it *is* working because I can upload gcode via my script and the cura plugin also works.

In my sleep deprived state I forgot to test with all the plugins disabled (plus I'm currently printing a bunch of stuff) so if you'd like I'll disable them once I'm done but could it be a plugin causing it?

I have read the FAQ.

foosel, 2017-02-13:
First of all, sorry for the late reply - only now somewhat back on my feet after a quite nasty cold.

I don't think it's a plugin's fault. I recently discovered though that the API key based admin session creation was missing a small step that made the API key get filtered out from the `/api/settings` results (will be fixed in 1.3.2), and that would cause exactly this to happen then. Additionally, the way things are currently implemented make it so that you get logged in as that API key admin account when you do a manual request with the API key in your browser. When that happens OctoPrint will display you as logged in as "_api" instead of your regular user name. Was/is this the case maybe? Because that would fully explain things.

> I don't know if it's related in any way but I got the update notification so let octoprint update itself and it refused to start back up again.

That was a bug in the init script that got fixed with 1.3.1, and on manual installs needed an update of the deployed init script to the updated bundled version. I added a heads-up to the release notes once I got a first report about this and remembered that this was necessary, I guess you did your update before I remembered to add that heads-up.

ntoff, 2017-02-13:
> Additionally, the way things are currently implemented make it so that you get logged in as that API key admin account when you do a manual request with the API key in your browser. When that [...]

I was just accessing it normally via octopi.local and I'm pretty sure I was logged in as the regular user but wasn't really paying too much attention to the username.

> I guess you did your update before I remembered to add that heads-up.

Must have, or I just missed the warning.

foosel, 2017-02-13:
Can you still reproduce it now (when actually logged in as your user account according to the corresponding entry in the navigation), or was it a one-time-thing?

ntoff, 2017-02-14:
hmm no, whatever it was (happened on two different pi's) seems ok now. Will keep an eye on it and if it happens again I'll try and remember to check the user name that it think I'm logged in as.

foosel, 2017-02-14:
@ntoff ok, marking it as unreproduced for now. If you don't see it again once 1.3.2 is rolled out, chances are high it was indeed the above mentioned issue that will be fixed in that upcoming version. Fix isn't pushed yet though, I'll try to remember to reference this ticket in the commit.

ntoff, 2017-02-14:
Just now I tried aaaaand the API key was empty. HA! I don't know any definite steps to reproduce it though. I wonder if it's because I cloned an already operational octoprint SD card, then changed the host name so the hostname is different but the API key is the same and Chrome is getting confused?

Either way leave it closed if it's only me being affected, it's probably one of the many weird things I'm doing at the moment that's just messing with the browser cache or something.

foosel, 2017-02-14:
Actually empty or "n/a"? And what did it say about the logged in user?

The host name should not make a difference here, there could be some caching issue though maybe.

ntoff, 2017-02-14:
n/a sorry and the user was just the admin user that it always is. I'm doing so many different things at the moment I don't know which one would be the one causing it.

What I have done though (that I can remember) is to clone my other working manual install to this SD card, change the host name (but not API key yet), change a bunch of stuff to do with the wireless adapter (too many random things to list, I'm having trouble getting this pi to be stable on the network but the n/a issue was on both).

I wonder if because the issue originally cropped up on my other pi and I've cloned it, whatever the random issue is has carried over. I might nuke everything and do a fresh install or something. Either way the reproduction steps are too random to figure out so I guess for now if there's nothing obvious just close it until someone else has the same issue since it's probably something I've done to mess it up.

robertwallace, 2017-02-14:
This happened today when I turned on my two raspis.  The API key in both were "n/a" .  Both were fine when I shut them down from the menu yesterday.

They were still configured, today - you can see the API key in the Access control settings (where I copied them for Slic3R) -- they were just not showing up in the API tab ("n/a").

foosel, 2017-02-14:
What did it show you logged in as in the upper right corner? Your user name or "_api"? What version of OctoPrint?

robertwallace, 2017-02-14:
My user name.  No "_api" extension.  Once I regenerated the key from the Access control tab (and Saved) the previous API key started showing up in the API tab.  But it seems to take either one when talking to my Slic3r install.  I've got the new one in the Slic3r tab, and the old one in the API tab, but it all seems to work.  Maybe after I shut them down and restart, API will be an issue again, but I'm in the middle of a long Octoprint-initiated job, so I'll be keeping them up for a while.

foosel, 2017-02-15:
> What version of OctoPrint?

^--- Important ;)

Also, which API key is it that's not set? Because you are mixing up user keys and the global API key. Is it this one (global API key):

[image]

or that one (user key, that one is also visible under Settings > Access Control for admins):

[image]

that says "n/a" at the moment? And which of those is it you are using in Slic3r? They are different and both will work, just behave differently. The global API key gives you permissions of a global system user with admin rights, the user key will give you exactly the same rights as if you were logged in with your username and password into the web interface.

I've just pushed a commit to the `maintenance` branch that should fix an issue that could lead to the *global API key* being reported as "n/a" if logged in as the aforementioned global system user instead of an actual user account. That's also why I asked what username was displayed as being logged in.

I would ask you @ntoff and you @robertwallace to [enable debug logging for the "octoprint.users" component](http://docs.octoprint.org/en/master/configuration/logging_yaml.html#logging-yaml) and then trying to reproduce the problem against the `maintenance` branch. If it still occurs, please provide `octoprint.log` - the increased log level might give us a hint as to what's going on, since it will log which exact user was logged in with which rights when a new session is created.

PS: I am freely sharing those screenshots with my keys here since they are associated with my development instance that never was, is or will be accessible outside of my LAN - if that was not the case I would obfuscate.

robertwallace, 2017-02-15:
I'm running Octoprint 1.3.1. , the latest version the interface has announced.

 I was logged in as "robertw", the name displayed on the upper right of the screen.  This is the only way I log into the web interface and it is the only userid I've added in access control.  However, it does have the "admin" checkbox checked in the Access Control, so that likely meets your definition of a global system user.

I wasn't aware of global vs. user when I first commented.   The global key was the one that was "n/a".  The one I generated, I now realize, was the user key - I did it from the Access Control tab and the little key icon to change password, not from the User Settings tab.  Instead of changing my user password, I generated a key - a user key, I understand, now, and copied that and used it to replace the original key in Slic3r.  When I later checked the Global key in the API tab, it had been populated (with the previous global key, I believe) and was no longer "n/a".  I tried them both in Slic3R and both worked. (as you mention.)  I didn't notice any change in privilege between the two keys, but didn't try to change any admin stuff.

I powered my raspi on this morning (after clean shutdown yesterday.) and the global API key field was populated, and was not "n/a" as it had been yesterday prior to my obtaining a user key -- all seems to be back to normal, now.   I will attempt to reproduce and log the problem, but I don't know what I may have done to cause it, so I'm unsure how I might reproduce it.

foosel, 2017-02-15:
Thank you, that's great info 👍 :) 

Also good to know that it in fact was the global key for you too that was missing in action (or rather, not displayed - "n/a" gets injected when the backend believes you don't have the rights to see the key, so it might be a caching thing after all). While debugging this earlier, or rather trying to since I can't reproduce it, I noticed that after login I don't immediately request new settings, so it might be that when you opened the settings (which triggers a refresh) you still saw the settings from the not-logged in state. In that case a little spinner should have been visible in the settings dialog title bar. That could also be what happened here, so when you stumble over it again, also check if that spinner is active and if the key shows up once the spinner vanishes again (= settings refreshed from backend).

I'll keep my eyes peeled over here too.

ntoff, 2017-02-15:
ok mine is the global user, I didn't even know there were two different API keys until now. I've always just given out the global api key (to my scripts / slicers, not given out to other people).

Everything still works, I can use the web interface just fine even with the n/a key so for me the issue seems purely visual (like it's just trying to read the wrong variable to display?).

My "user settings" api key says n/a and hitting the regenerate button doesn't do anything.

>so it might be that when you opened the settings (which triggers a refresh) you still saw the settings from the not-logged in state. In that case a little spinner should have been visible in the settings dialog title bar.

Yes, I do see that little spinner. This might be dangerous but should I just give you access to my spare pi? It's not plugged into a printer because the wifi dongle on it is broken and there's no ethernet in the print room but I can plug it into the ethernet and let you have at it with the virtual printer.

Password is 1 2 3 4 5 (little spaceballs reference there for humour)

foosel, 2017-02-15:
I actually did get that reference :D It's been ages since I last watched that movie, I need to change that.

Taking a look myself might actually be helpful. You can shoot me an email with the connection data at gina -at- octoprint.org, no need to post that stuff in the public.

ntoff, 2017-02-16:
ok email sent, if you want a real time chit chat (like if the pi goes down or something and you need it rebooted) feel free to message me in hangouts or something.

robertwallace, 2017-02-24:
I've since found that when the API key shows up as n/a, if I just click the Save button (on the API screen or, I think, any settings screen), it repopulates and is there for the rest of time I have Octoprint running.  But it's empty the next time Octoprint starts up from an Pi startup.  So maybe all that is needed is a refresh of the form's field.

foosel, 2017-03-17:
I still haven't been able to reproduce this behaviour, but I've changed the whole management approach for the general API key for 1.3.3. It will have a "regenerate" button there and behave more in line with the user API key. Maybe while changing that, whatever is causing this for you also got nuked. Or the commit up there (e8349d6) that made it into 1.3.2 already does it.

It sounds like a caching issue, but my attempts to reproduce that have been completely unsuccessful so far, even on @ntoff's machine

foosel, 2019-11-05:
Since I haven't heard of any further reproductions of this since early 2017 I'm assuming this fixed.

closed 2019-11-05 by foosel