OctoPrint/OctoPrint #1754: Octoprint Continuously Looping with no Printer Functionality

Myko7, 2017-02-06:
What were you doing?
Updated to OctoPrint 1.3.1 version. Connecting to my printer and proceeding to control temperature and position to home.

What did you expect to happen?
Expecting the nozzle to heat up accordingly and return to home.

What happened instead?
The LCD on the printer continuously restarts without showing the temperature details as normal. Web interface buttons does not control the printer whatsoever, although the state denotes "operational" mode. The web interface still "works" but all printer related functions don't.

Branch & Commit or Version of OctoPrint
1.3.1

Printer model & used firmware incl. version
Monoprice Maker Select i3 V2

Browser and Version of Browser, Operating System running Browser
Firefox 51.0.1 (32-bit)

Link to octoprint.log

https://gist.github.com/Myko7/0791448111510220a4a62436baefdb8e

Continuously looping the following:
"Recv: Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
Recv: Free RAM:10732
Recv: wait
Recv: start"

Screenshot(s) showing the problem:

http://imgur.com/jseUlVc

I have read the FAQ.

Myko7, 2017-02-06:
Here's a picture of the connection part of the web interface!

http://imgur.com/LjLKZnW

foosel, 2017-02-10:
You did not provide octoprint.log (the linked file is only the terminal contents). That log file is mandatory for any further analysis.

It's in fact not OctoPrint that's looping here but your printer that keeps endlessly resetting. Does this happen with other hosts? Did this use to work with OctoPrint? If so, with what version and what apart from the version of OctoPrint did change? What happens when you unplug the printer's USB cable when it does this, does it keep restarting? Have you tried a different USB cable?

Myko7, 2017-02-13:
You're right. My cousin accidentally shorted my printer without my knowledge, but the printer is now fixed and Octoprint is running as expected! Thanks

closed 2017-02-13 by Myko7