OctoPrint/OctoPrint #1755: Possible memory leak with big objects

Foxandxss, 2017-02-06:
#### What were you doing?

I am printing a 31.9mb object. When I try to see the live preview "Gcode Viewer" it starts analyzing the piece and then goes up to 2GB of ram to then crash the tab.

#### What did you expect to happen?

Show the proper gcode in live preview.

#### What happened instead?

[image]

#### Branch & Commit or Version of OctoPrint

1.3.1 master

#### Printer model & used firmware incl. version

Anet A8, Skynet3d (fork of Marlin).

#### Browser and Version of Browser, Operating System running Browser

Latest Chrome stable, Windows 10.

#### Link to octoprint.log

https://gist.github.com/Foxandxss/ffc9f85532095a97d5c12181b6520eeb]

#### Link to contents of terminal tab or serial.log

https://gist.github.com/Foxandxss/ffc9f85532095a97d5c12181b6520eeb

#### Link to contents of Javascript console in the browser

Nothing relevant here.

----

I printed hundred of pieces, this is the first one giving the issue and perhaps the bigger one in terms of gcode.

foosel, 2017-02-10:
Please provide the gcode file to reproduce this with.

Foxandxss, 2017-02-11:
https://dl.dropboxusercontent.com/u/10283624/Hairy_Lion.gcode

There you have it. It is the classic hairy lion.

All things considered, with EDGE it was working nice, was only Chrome.

foosel, 2017-02-13:
Hm, nope, can't reproduce with that over here. Memory usage also doesn't go to 2GB for me, not even close, currently at a pretty stable 600MB, so no leak either as far as I can see.

Is this reliably reproducible with this file for you or an on-off-kinda-thing? When it crashes, does it always crash during analysis (after it says "Loading" and before it says "Analyzed")? Or does it get to "Analyzed" and then crashes when attempting to perform the actual visualization? Since I can't reproduce the behaviour with Chrome over here (also on Win10), it might also be worth a shot - if you can reproduce it reliably - to see if a fresh profile without any extensions and such shows the same behaviour, just to rule out anything like that from being the cause here.

In general, the GCODE viewer does in fact consume a lot of memory (it has to parse the whole file, unpack the information contained in all lines therein and store that information in memory in a way it can easily be accessed when quickly scrolling through the individual layers and such, and the information contained in such a file when parsed is more than the size of the GCODE lines making it up thanks to additional context information being required). Which is why larger files (with a separate threshold for desktop and mobile devices) will display a "are you really sure" first before automatically visualizing. The default value for desktop is 20MB, so your Hairy Lion GCODE is just below that. You can lower the threshold via Settings > GCODE Visualizer.

There *does* exist a bug for "spiralized" models (see #1235), but your model doesn't fit the bill so that shouldn't be what's happening here.

Foxandxss, 2017-02-18:
I will run this print again probably tomorrow before I change this printer. My octoprint is open to the network, so perhaps you could try yourself and see if it is only me or not.

foosel, 2017-02-20:
If you shoot me the URL via mail, I can take a look: gina -at- octoprint.org. Don't post it here. And in general make sure to properly secure your instance please (additional access control outside of the actual machine running OctoPrint e.g. through htaccess, ideally SSH tunnel or VPN) - OctoPrint is a print server, not a network security solution.

foosel, 2017-03-08:
Closing as suggested by OP via mail since I couldn't reproduce it even when connected to their instance with my own browsers.

closed 2017-03-08 by foosel