OctoPrint/OctoPrint #1757: Error: Offline: Too many consecutive timeouts, printer still connected and alive?

jaccovh2, 2017-02-06:
Today my 3D printer was busy printing quite a big 3D object, only after 7 hours my printer stopped and according to OctoPrint there where too many consecutive timeouts. This already happend to me a couple of times, even during smaller prints. Any idea how this could happen?

#### What were you doing?
Printing a large 3d model, printing should have taken about 10 hours.

#### What did you expect to happen?
After 10 hours a successful 3D printed object.

#### What happened instead?
After 7 hours printer suddenly stopped and according to OctoPrint:
> Offline: Too many consecutive timeouts, printer still connected and alive?

#### Branch & Commit or Version of OctoPrint
Version: 1.3.1 (master branch)

#### Printer model & used firmware incl. version
TEVO Tarantula I3 with a Raspberry Pi model 3 running OctoPi

#### Browser and Version of Browser, Operating System running Browser
Safari on macOS

#### Link to octoprint.log
OctoPrint log: https://gist.github.com/jaccovh2/a65df3826e22022440256ef3e2243fe2

#### Link to contents of terminal tab or serial.log
OctoPi terminal: https://gist.github.com/jaccovh2/75a4cb035740d671839bda9d312be122

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s) showing the problem:
[image]
Screenshot showing some of the serial connection settings:
[image]

I have read the FAQ.

foosel, 2017-02-10:
Your printer simply stopped responding: 

```
Send: N304729 G1 X39.296 Y55.678 E23648.53982*89
Recv: ok
Send: N304730 G1 X39.543 Y55.662 E23648.55217*84
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N304731 M105*21
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N304732 M105*22
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N304733 M105*23
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N304734 M105*16
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N304735 M105*17
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Printing' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor
```

OctoPrint can't talk to your printer when it doesn't reply, and it can do nothing but abort when that status persists.

Try a different USB cable between your printer and your OctoPrint server, with a ferrite bead and as short as possible. Make sure it doesn't run next to motor wires. Make sure the PSU for both your Pi (if you are using that for OctoPrint) and your printer are solid and actually producing the rated voltage and current (the Pi's red LED should light up solid). Mobile phone chargers are not suitable power supplies for a raspberry pi.

This is not a bug in OctoPrint, this is an issue with the communication to your printer outside of OctoPrint's control. I can't do anything about it other than tell you the above, sorry.

closed 2017-02-10 by foosel

Erutan409, 2017-02-18:
@foosel In reference to my [previous comment](https://github.com/foosel/OctoPrint/issues/1506#issuecomment-279436076) - I just witnessed another disconnect with my printer, unfortunately.  I just want to make sure I'm clear on what you're explaining is that it's not necessarily an issue with OctoPrint, but with the printer communicating with the Raspberry Pi, itself?

If I'm able to print via USB from Simplify3D without any issues, is it safe to assume that the aforementioned slicing software is somehow accounting for these same kinds of communication issues that we're seeing from/in OctoPrint's terminal?

I haven't gone looking for the error message that's outlined in this ticket (in Simplify3D's terminal), because I haven't had a failed print using Simplify3D.  So, I can't confirm or deny that it IS happening with all USB printing methods.  I was also printing with the included MatterControl software for my Robo 3D R1+.  That never had issues printing via USB, either.

So, based on general observation (with my limited knowledge on the communication protocol(s)), I'm leaning on this being an issue with either, Marlin, the Raspberry Pi, OctoPrint, or a combination of all 3; although Marlin seems to be working correctly/cooperating with PC slicing software.

I'm just trying to ascertain which road to travel down to figure this out.  I'm willing to do any further troubleshooting to help sort this out for myself and others, since, it's a widely reported issue.  Is there anything that I could try doing that IS within OctoPrint's control, like forcing the software to wait longer for the response?

I hate to beat this dead horse, because based on what I just explained, I don't see why this horse is dead.  Unless, it's just an issue with USB communication on the RP, I could try running OctoPrint on something else and just be done with it.

ctgreybeard, 2017-02-18:
@Erutan409 Let me jump in here with a question. There is a red LED on the Pi3, it's with the green LED on the opposite end from the USB, etc.  Is that LED on steady 100% of the time? No flickers or flashes?

I ask because on my Pi3 that corresponds to adequate power. It will flicker if the Pi has to throttle due to low power availability. Until I upgraded my power supply to a 5.25v supply I would get flickers and the Pi would sometimes misbehave.

Erutan409, 2017-02-19:
@ctgreybeard My red LED is constant/steady 100% of the time, that I can reasonably observe.  No flickering that I can detect.  With and without both the 3D printer's Arduino board and my C290 Logitech webcam plugged into the USB ports on the RP, I can measure a perfectly steady 5.2 volts by using PP3 & PP7 on the bottom of the board.

From what I can tell, I also have the correct adapter (that was also marketed toward the RP, when I picked up both the power supply and said RP from Fry's Electronics (of which I have done, twice):

[image]

I also want to iterate that I am by no means intending to come across as terse from an emotional standpoint.  I only wish to give direct feedback to those, such as yourself, who are willing to assist with troubleshooting.

Thank you! 👍

Erutan409, 2017-02-19:
@ctgreybeard I've been toying with the idea of running the Arduino board connection through a powered USB hub to rule out any potential issues with consistent electrical current via the USB connection.  Any thoughts on that on whether or not that could make a difference?

ctgreybeard, 2017-02-19:
Well, I only stepped in here because I had power problems that I was unaware of. I didn't realize the flickering red LED was an indicator of trouble, I thought it was an activity indicator. Once I installed a power connection very similar to what you have then my troubles went away. It sounds and looks like you are OK in the power domain. I don't think a powered hub would help but also wouldn't hurt as long as it's capable of delivering the current.

The only other area that I can recall where this problem has been reported is in the USB cable from the Pi to the printer. I would ensure that it is a quality cable and kept away from any power-handling cables. Keep that in mind for internally to the electronics enclosure of the printer itself too. My Prusa i3 MK2 is a mess-o-wires inside the box and it took me a while to get them sorted out to keep the signal and bed/extruder power as far from each other as possible. For me it was manifested as weird characters on the LCD but I suspect the USB could also experience interference.

Erutan409, 2017-02-19:
Makes sense.  So, I just recently shortened the USB cable to the RP, which is what I noted in my referenced comment above.  My only reservations about signal noise from electrical cables is that I'm still succeeding 100% of the time while printing from USB on my PC slicing software.

I think I'm going to start skimming Marlin issues/forums to see if they shed any light on this.

ctgreybeard, 2017-02-19:
Yeah, I've seen others with similar results between to Pi and their laptop/whatever. Could it be that the line drivers on the Pi aren't as robust as those on the laptop? Or, perhaps, the cable position is slightly different? I don't know. I do know that this is not an unusual problem but it doesn't seem to have an easily identifiable solution.

nophead, 2017-02-19:
Since USB has its own retry mechanism at the link level it usually either
works perfectly or it disconnects. I have never known it to simply stop
receiving, or get corrupt data (I have seen corrupt data but the corruption
happens after the USB link) . So it looks like your printer stops
communicating.

If your printer is using 5V from the USB cable that is generally a bad
idea. The 5V USB VBUS is only suitable for regulating down to 3.3V. It
doesn't stay in spec for 5V devices but unfortunately a lot of people don't
understand that.

Erutan409, 2017-02-19:
@nophead 

> If your printer is using 5V from the USB cable that is generally a bad
idea. The 5V USB VBUS is only suitable for regulating down to 3.3V. It
doesn't stay in spec for 5V devices but unfortunately a lot of people don't
understand that.

What context are you specifically referring to?  For clarification, the Arduino in my printer shouldn't be relying on the raspberry pi for its main source of power.

nophead, 2017-02-20:
You can run an Arduino from its USB connection or you can apply power to the power jack. The latter is much better because it has a 5V regulator and so gets a stable 5V rail.

foosel, 2017-02-20:
> I just want to make sure I'm clear on what you're explaining is that it's not necessarily an issue with OctoPrint, but with the printer communicating with the Raspberry Pi, itself?

Yes, that's my point. It depends on the exact error you are seeing though on the disconnect. There are also things that can cause a disconnect that are triggered by errors thrown by the firmware.

> If I'm able to print via USB from Simplify3D without any issues, is it safe to assume that the aforementioned slicing software is somehow accounting for these same kinds of communication issues that we're seeing from/in OctoPrint's terminal?

It can't. But the hardware you are running S3D on might account for these issues. The Pi is not a full fledged PC. It has its issues with power (as mentioned above), it is somewhat more picky when it comes to connected hardware (not talking about the pickyness of Linux itself here or rather the lack of vendor support). It doesn't provide the same amount of constant power output on its USB ports that you would get when connecting to an actual PC, so THAT might be a factor too depending on your printer electronics.

> the Arduino in my printer shouldn't be relying on the raspberry pi for its main source of power.

Shouldn't. I've also seen reports of people where that turned out to be the case after all.

Thing is, as much as I wish there was a simple fix to this kind of issues that I could put into OctoPrint, there simply is none that I'm aware of. Either I can connect to the printer via serial and said serial connection stays open and stable until I close it. In that case I can work fine. Or the serial connection randomly crashes or stalls or similar stuff. In that case I can do nothing about it. I can't reconnect automatically - if the printer happens to print from SD in that moment I'd disrupt the print because a huge amount of controllers reset on serial connect. That would also make the firmware loose its position, leading to all kinds of followup problems. Plus I'd potentially mask an issue that the printer's operator should take care of. It's simply a case of not having any options to work around this on my end - I can't fix these kinds of issues in software, as much as I'd like to.

Erutan409, 2017-02-20:
All good to know.  I'll keep tinkering on my end.  Thank you, @foosel.