OctoPrint/OctoPrint #1758: Error on Start Server

developer2013, 2017-02-07:
Hello, 

i have installed OctoPrint and now will run it with command: ~/OctoPrint/venv/bin/octoprint

i get the follwing error: 

`Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.1', 'console_scripts', 'octoprint')()
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/OctoPrint-1.3.1-py3.4.egg/octoprint/__init__.py", line 398, in main
    from octoprint.cli import octo
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/OctoPrint-1.3.1-py3.4.egg/octoprint/cli/__init__.py", line 134, in <module>
    from .dev import dev_commands
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/OctoPrint-1.3.1-py3.4.egg/octoprint/cli/dev.py", line 232, in <module>
    @dev_commands.group(name="dev", cls=OctoPrintDevelCommands)
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/click-6.2-py3.4.egg/click/core.py", line 1169, in decorator
    cmd = group(*args, **kwargs)(f)
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/click-6.2-py3.4.egg/click/decorators.py", line 115, in decorator
    cmd = _make_command(f, name, attrs, cls)
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/click-6.2-py3.4.egg/click/decorators.py", line 89, in _make_command
    callback=f, params=params, **attrs)
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/OctoPrint-1.3.1-py3.4.egg/octoprint/cli/dev.py", line 24, in __init__
    from octoprint.util.commandline import CommandlineCaller
  File "/home/pi/OctoPrint/venv/lib/python3.4/site-packages/OctoPrint-1.3.1-py3.4.egg/octoprint/util/__init__.py", line 8
    from __future__ import absolute_import, division, print_function
    ^
SyntaxError: from __future__ imports must occur at the beginning of the file
`

has anyone an idea? 

regards
Carsten

FHeilmann, 2017-02-07:
Your log indicates that you are using Python3 to run Octoprint. Octoprint is not compatible with python3 yet.

developer2013, 2017-02-07:
Hello Florian, how can i start OctoPrint with Pyhton2 oder Pyhthon 2.7?

FHeilmann, 2017-02-07:
Please provide some more information about the system you are running octoprint on.

Also provide the output of the following three commands:

`which python`
`which python2`
`source ~/OctoPrint/venv/bin/activate & which python & deactivate`

developer2013, 2017-02-07:
in which python he shows: /usr/bin/python
in which python he shows: /usr/bin/python2

te
[1] 15631
[2] 15632
/usr/bin/python
-bash: deactivate: Kommando nicht gefunden.
[2]+  Fertig                  which python

?

closed 2017-02-22 by GitIssueBot