OctoPrint/OctoPrint #1763: [Request] WLAN / LAN configuration in settings

BigBox3D, 2017-02-10:
It would be very helpful to be able to configure the LAN/WLAN in the settings tab from within OctoPrint.

Options such as settings a static LAN IP, entering WLAN SSID and Passwords.

I understand there is the octopi-network.txt file however it would be favourable to have a GUI within OctoPrint to do the same.

Salandora, 2017-02-10:
#367 what about this one? (down at the end)
Atleast you can use it to configure wlan with it.
Actually I have no Idea in what state the project is, 
but I believe @foosel can say if it is in a workable state.

foosel, 2017-02-10:
The problem with netconnectd is that while the service itself works great (and is used by some projects who are in control of their hardware), general roll out with OctoPi proofed horribly troublesome thanks to the myriad of usb network dongles out there, making it next to impossible to get access point mode working reliably. This might be better achievable with the standardised hardware of the pi3 now, but everything else is a horrible mess to get working via plug and play and more importantly has unclear licensing status (since for some chipsets custom drivers would need to be bundled for ap mode).

Now, without bothering with access point stuff it would be possible to build something to configure network settings within OctoPrint (with an intermediary service so that no superuser rights would be necessary) or just use netconnectd as is, but in that case I'm having a bit of an issue to picture the actual situation in which you'd use that. Either you can already access your OctoPrint instance via a network - then you obviously already configured it and don't need to do that anymore. Or you can't, but then you can't access the interface to perform initial configuration either (assuming a headless mode here, which should be the default). So it doesn't make sense without an access point mode. I could picture wanting to configure a static IP instead of the one assigned by dhcp - but that could also trivially be done in the dhcp server and if not there's bonjour and upnp in place.

If you control the hardware of your devices though, by all means take a look at the aforementioned netconnectd, that should have you covered:

  * https://github.com/foosel/netconnectd
  * https://github.com/OctoPrint/OctoPrint-Netconnectd

foosel, 2017-03-17:
@BigBox3D did you receive that comment above? :)

John-Mc, 2017-04-09:
Would the method to automatically create a hotspot if no known SSIDs are found that @kennethjiang is working on help with this? Scroll down about 1/3 of the page on [Painless Wi-Fi for OctoPrint](https://medium.com/@kennethjiang/painless-wi-fi-for-octoprint-4e6b68005400) to find the hotspot part. 

I've tested the auto-hotspot out fairly thoroughly on my Pi3 running OctoPi 0.14rc1 and found it stable. (The only issue of which I'm aware is that if an SSID entered in octopi-wpa-supplicant.txt is discovered, but you have the wrong password entered, it will not create the wifi hotspot. It sees the known SSID and stops. Kenneth is aware of this, but has not had time to address it yet.)

kennethjiang, 2017-04-09:
@John-Mc I fixed the problem in the situation of wrong password and updated my blog.

Meanwhile @foosel 's netconnectd seems to be a more tested solution to this issue (I wish I had discovered it earlier). If even netconnectd didn't pass the test of reality of the messy wifi world, it'll take much more effort to codify my blog into a stable solution.

foosel, 2017-04-10:
As mentioned, the problem is with the fragmented hardware we are dealing with. Exclusively targeting the Pi3 would allow for something like "fire up a hotspot", but that would leave out at least half of the population. It might still make sense to have *something*, but I worry a bit about the communication of compatibility here - people don't read, "AP if Pi3 or PiZeroW" will get overlooked. In any case, this is something that needs to be solved one layer beneath OctoPrint, possibly combined through plugins. That is what netconnectd attempts to do, minus a solution for hardware fragmentation. 

I also know I still have to reply on the relevant ticket on the OctoPi tracker - I simply haven't yet gotten around to that.