OctoPrint/OctoPrint #1765: Bug: Octoprint displays wrong file size and estimates wrong ETL

wschadow, 2017-02-10:
Hi,

attached please find a screen shot of the issue. In the file list you can see that the file has 21.6 MB. In the print dialog it only shows 7.9MB and it printed 8.0MB (?!) of that file. The ETL was displayed wrong; it seems to have taken only 7.9MB into account for the calculation. It displayed a time until the 7.9MB were reached, now it's still "stabilizing", but at least it keeps printing :)

Cheers,
Wolfgang

[image]

wschadow, 2017-02-10:
#### What were you doing?

printing a gcode

#### What did you expect to happen?

print the gcode, and diplay correct information

#### What happened instead?

wrong information displayed: n the file list you can see that the file has 21.6 MB. In the print dialog it only shows 7.9MB and it printed 8.0MB (?!) of that file. The ETL was displayed wrong; it seems to have taken only 7.9MB into account for the calculation. It displayed a time until the 7.9MB were reached, now it's still "stabilizing", but at least it keeps printing :)

#### Branch & Commit or Version of OctoPrint

    Version: 1.3.1 (master branch

#### Printer model & used firmware incl. version

Prusa I3 MK2

#### Browser and Version of Browser, Operating System running Browser

Firefox, Win10

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-02-10:
Log files please, they are not optional. Provide *all* requested data. It's impossible to debug anything without that.

Also: did you by any chance do the upload of that file by watched folder?

In general OctoPrint takes the file size it displays there from the file system.

wschadow, 2017-02-10:
the print is not finished, could it that the content is buffered? I'll provide that, once the print is finished
[octoprint.zip](https://github.com/foosel/OctoPrint/files/767965/octoprint.zip)

foosel, 2017-02-10:
No, the content is not buffered but the file size is only read at the start of the print. 

Again, how did you upload that file? Web interface, API (eg through Slic3r), watched folder?

foosel, 2017-02-10:
Ah, log says watched folder. This is a duplicate of #1760 then :)

closed 2017-02-10 by foosel

wschadow, 2017-02-10:
I had uploaded the log file already.

 

The file was uploaded from Simplify3d into the watched folder. BTW the correct file size is in the filelist (see the screenshot)

wschadow, 2017-02-10:
I don’t think it’s a duplicate. I uploaded the file only once. Second the correct filesize is displayed in the file list, but not when it’s printed. You judge :)

foosel, 2017-02-10:
Let me put it this way, something's up there with the watched folder with regards to files uploaded through it and their file size as reflected in the job data (which is read at a different point in time than the data in the file list). The original ticket ran into this issue when uploading a second version, you just proofed that the issue also manifests with a first version. But both problems are caused by the same piece of code (which I'll have to go through with a very fine toothed comb) and in all likelihood also the same core issue, the symptoms are just too similar for this to be two isolated issues.

wschadow, 2017-02-10:
Ok, fine with me. At least you have enough information and know how to fix it ;)

foosel, 2017-02-10:
I hope so, if not (eg if I'm not able to reproduce it based on the information I now have) you might get a ping though from the other ticket to provide more details on your exact setup and workflow ;)

wschadow, 2017-02-10:
Of course :)

nophead, 2017-02-11:
> auth/AYgaaA-jofhynHOh0PJcO5A64r3vnddAks5rbPQrgaJpZM4L97Lb> . <

How does OctoPrint cope with a slowly written file in the watched folder?
I.e. when output from a slicer how does it decide it is complete? Should
files only be moved into there?

foosel, 2017-02-11:
Files should only be moved from watched to uploads and only when that completes fully should the event be fired that refreshes the file list and also triggers any auto select or print functionality. But I could imagine that something is happening there that's either firing that too early or there is some form of buffering going on after all even though it shouldn't which allows the file to be selected for printing before it is fully written. The problem here is that since I never expected a file to still be written when it is being selected, I'm only reading the file size during initial selection and not during printing (which also has performance reasons - it costs cycles and it should usually not be necessary). So if the file grows after selection stuff gets out of whack.

I'll have to solve this by first figuring out what is different between the process in the handling of the watched folder and the regular upload (because something is different here or things like the above would happen all the time with regular upload too), but also making the file size tracking in the job handling more resilient against file size changes.

wschadow, 2017-02-11:
Is there no indication that a file is still open? One could imagine a scenario where it takes a long time to write a file for whatever reason. In this case you need to figure out the “right” time to move the file from the watched folder.

foosel, 2017-02-11:
The problem here isn't that the file was moved too early from the watched folder (or your print wouldn't have continued but stopped in the middle). The problem is rather the move from watched folder to regular file storage/uploads folder, or maybe even just the timing of the "oh, look, new file" signal during that move.

But I won't get around to look into this until next week, and guessing at possible causes without a close look at the code is pretty moot ;)

wschadow, 2017-02-11:
Hm, the two sentences sound contradictory to me, maybe too early in the morning :)

 

In my “case” I was printing some other file while I uploaded a new one (once!) to the watched folder. In the file list it showed up correctly. At that time it was moved from the watched folder to the upload folder I think. Later, when I started printing the size did not show up in current job display …

nophead, 2017-02-11:
> AYgaaARgI7mmkDBfECxOpxYK2f3oYYgdks5rbW6sgaJpZM4L97Lb> . <

When the watched and uploads folders are on different file systems then a
move becomes a copy and delete, so I can imagine the bug might be caused by
delayed write buffering. I have certainly run into race conditions on
Windows file systems with Python when I started using SSD. Here is an
example of my horrible workaround to rmtree() returning before it has
finished removing files.

            shutil.rmtree(target_dir)   #if making the BOM clear the
directory first
            sleep(0.1)
            os.makedirs(target_dir)

But I am still curious in how you know the file in the watched folder is
complete when it can be written arbitrarily slowly and can be arbitrarily
long.