OctoPrint/OctoPrint #1766: Cancel GCODE Script not being sent

markwal, 2017-02-11:
#### What were you doing?

Steps:
1. Create a GCODE Script for Cancel via Settings->GCODE Scripts (Say G28\nM104\nM140 S0\n)
2. Start a print
3. Click Cancel

#### What did you expect to happen?

G28
M104 S0
M140 S0

to get sent after the printer runs through it's queue and says "ok"

#### What happened instead?

OctoPrint stopped sending, returned mode to Operational and only M105's after that.

#### Branch & Commit or Version of OctoPrint

master 1.3.1

#### Printer model & used firmware incl. version

Polygon 2060 running on a Replicape using Redeem develop branch (commit 0dc79293)

#### Link to octoprint.log

https://gist.github.com/markwal/a2c5971b84e0fb7792109319c7a455d8

I tried this on my non-GPX printer so I don't think it related to my plugin, but I included the log in case it is one of the other plugins I'm using.  I haven't looked into the issue cause yet.

I have read the FAQ.

foosel, 2017-02-11:
Can you also provide a serial.log? The cancel script is triggered on state transition and the state are logged in there. I wonder if that might give us a clue.

I certainly have made heavy use of my modified cancel script over the past months, so I wonder why it doesn't appear to trigger for you.

markwal, 2017-02-11:
https://gist.github.com/markwal/5af77cfb09fe637ed8d3188983fc7054

MoonshineSG, 2017-02-14:
Just a side note, that having a G28 at cancel can be a bad, bad idea! Chances are your hotend will end up crashing into the partial print.  (unless your printer model is such that this can never happen)

markwal, 2017-02-14:
Yeah. I know.  But in my case emergency stop is the power button and this particular model is a delta and G28 is up in Z until hit all endstops so most of the time it is what I want.

MoonshineSG, 2017-02-14:
In that case forget what I said. :)

foosel, 2017-02-21:
I still can't reproduce this :/

markwal, 2017-02-25:
I put logging in standard.py and the callback that fires the cancel event never gets called.  I suspect it is because regex_position.search never matches and so self._record_cancel_position is permanently true and _cancel_preparation_done never gets called. 

The Redeem firmware answer to M114 looks like this:

Send: M114
Recv: ok
Recv: ok C: A:0.0 mm B:0.0 mm C:0.0 mm E:0.0 mm H:0.0 mm X:0.0 mm Y:0.0 mm Z:30.0 mm

markwal, 2017-02-25:
Yeah. those "mm" in there makes it not match the regex.

markwal, 2017-02-26:
Ah.  It's not just the mm units, but also the order.  OctoPrint wants X Y Z E.  Looks like you can put anything you want before or after that.

markwal, 2017-02-26:
PR into Redeem's repo here: https://bitbucket.org/intelligentagent/redeem/pull-requests/102/make-redeem-m114-response-compatible-with/diff

foosel, 2017-02-26:
Perfect, thank you. I guess we still need a solution though to somehow still send the cancel gcode, even if the position could not be recorded. At least as long as it doesn't contain the placeholders to inject that position that is. Tricky...