OctoPrint/OctoPrint #1767: Timelapse Mode ZChange Does not save settings

totalitarian, 2017-02-12:
#### What were you doing?

Trying to save the default timelapse mode as zchange

#### What did you expect to happen?

Restarting the server - zchange would stay as the current timelapse mode
#### What happened instead?

Timelapse mode reverted to 'timed'
#### Branch & Commit or Version of OctoPrint

1.3.0 (master branch)

#### Printer model & used firmware incl. version

FFCP 2016

#### Browser and Version of Browser, Operating System running Browser

Chome

#### Link to octoprint.log

on request

#### Link to contents of terminal tab or serial.log

on request
#### Link to contents of Javascript console in the browser

on request

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2017-02-13:
Cannot reproduce. You are also running an old version, 1.3.1 is current. Update, then reproduce please and provide *all required information*.

> on request

Consider the ticket template itself the request. 

Imagine me having to run after this information for each and every single ticket from OctoPrint's thousands of users that gets opened here. There's a reason why this information is mandatory for tickets.

totalitarian, 2017-02-13:
Thanks, the update to 1.3.1 fixed the issue.

closed 2017-02-13 by totalitarian