OctoPrint/OctoPrint #1770: Initial serial connection needs a connect/disconnect/connect cycle

skorokithakis, 2017-02-13:
I (and many, many other people I know) have the following problem on a Wanhao i3:

When connecting to the printer, after pressing Connect, the connection takes forever (it never completes). Pressing Disconnect and the Connect again connects to the printer right away.

I went to Octoprint thinking "this'll be easy to reproduce endlessly", as it happens every single time, but I can no longer reproduce this with 1.3.1, it connects immediately the first time. I'm going to file this issue anyway, because it's a widespread problem, but keep in mind that it *might* be fixed with 1.3.1.

skorokithakis, 2017-02-13:
Hmm, here's something odd, which I believe might have something to do with the issue. I reproduced the problem, but when trying to copy the log, I can't copy past some point:

~~~
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: 
~~~

However, this is what I see:

[image]

I think the printer might be sending some odd character that confuses OctoPrint? I'm not sure if the "Recv" lines are strictly what the printer is sending (i.e. if the printer is sending "wait" verbatim) or if that's OctoPrint's logging, but I think I should print hex chars instead so we can figure it out.

If you can tell me which file handles logging to that window, I can go change it to output hex instead, and report back.

foosel, 2017-02-13:
Please provide a fully filled out ticket. As stated in the other ticket I need logs, not excerpts from logs, information on versions, hardware, firmware etc etc :) You'll also need to enable `serial.log` and provide that from a connection attempt. And also `octoprint.log`. All information on where to find *and enable* the requested logs are in the linked Contribution Guidelines. Please read them and provide a fully filled out ticket.

Don't take this the wrong way, but it's really really frustrating getting "Hey, something doesn't work here" issues all the time but without the information I desperately need to be able to help in any way and then having to run after that. *Especially* when it's a case of "and many others". It's exhausting and it sometimes makes me lash out and I don't like that :(

ntoff, 2017-02-19:
>If you can tell me which file handles logging to that window, I can go change it to output hex instead, and report back.

@skorokithakis settings -> logs -> serial.log but first you need to enable it in the "serial connection" section.

You shouldn't  need to change anything to output as hex. Enable the serial log, reproduce your issue, then go into logs and download the log. It should download to your own computer which you can then just attach  here.

Do you have anything set to auto? If you do, try setting the port and baud rate manually and see if it connects ok.

skorokithakis, 2017-02-19:
I have attached two logfiles. One, `failed-serial.log` is the first attempt (that never completes), and `serial.log` is the second, successful attempt. You can see in `failed-serial.log` that it contains the following line:

~~~
2017-02-19 13:43:37,835 - Recv: \00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00ok
~~~

There's a long string of received null bytes before the OK, which confuses OctoPrint, I think.

[serials.zip](https://github.com/foosel/OctoPrint/files/785875/serials.zip)

skorokithakis, 2017-02-19:
Removing the null bytes in `_readline()` does not seem to fix this, for what it's worth.

ntoff, 2017-02-19:
I wonder if repetier firmware is assuming binary communication?

https://github.com/repetier/Repetier-Firmware/blob/master/repetier%20communication%20protocol.txt

foosel, 2017-02-20:
As far as I understand that protocol and its implementation, it's only for transfering GCODE to the printer, the responses are unmodified plain ASCII. Might be mistaken though.

In any case, that null byte prefix is indeed the issue because OctoPrint checks for `ok` on the start of a line (otherwise something like `SD card ok` or `Opened file file_ok.gco` or similar would cause issues). It also looks like the controller is not resetting here, so the usual fallback of "If you see `start`, try the handshake again" doesn't trigger.

I've pushed a commit to `maintenance` that will a) strip null bytes and b) also make sure to interpret `wait` during connection as a case of "oops, apparently we missed an `ok`", so that even if we get something entirely messed up like `\0\0\0o\r\n\0\0\0k\r\n` as acknowledgement to our very first `M110` it should hopefully still get things running. I tested this against the built-in virtual printer which I also adjusted to behave like what we saw in the serial log file up there and in that case it fixes things, but it would be great to get a report back @skorokithakis if it also solves the problem with an actual printer.

As a small side-note, that whole adjustment is only to get around any kind of weird garbage on the line during initial connect though (and I actually remember seeing something similar on early versions of the Hephestos 2 firmware, were the bootloader was doing some odd things, so that's not an isolated case even though it still strikes me as odd).

skorokithakis, 2017-02-20:
I tried stripping null bytes already, I'm afraid it made no difference...

foosel, 2017-02-20:
It definitely made here, but as mentioned, I also did an additional step to even recover from the case that stripping the null bytes isn't sufficient (and I just pushed another commit for that since I forgot something in that additional step)

Here's what the connection to the virtual printer with simulated garbage now looks like:

```
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N1 M105*38
Recv: ok
Send: N0 M110 N0*125
Recv: T:0.00 /0.00 B:1.00 /1.00 @:64
Recv: ok
```

and here's what now happens even without stripping the null bytes thanks to the second fallback workaround:

```
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ok
Recv: wait
Changing monitoring state from 'Connecting' to 'Operational'
Send: N1 M105*38
Recv: ok
Send: N0 M110 N0*125
Recv: T:0.00 /0.00 B:1.00 /1.00 @:64
Recv: ok
```

(simulated Repetier Firmware behaviour, hence the `ok` comes before the command output).

For reference, this is without the changes:

```
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ok
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Connection closed, closing down monitor
Changing monitoring state from 'Connecting' to 'Offline'
```

skorokithakis, 2017-02-20:
Ah, alright. Can you give me the list of commits to apply to my branch so I can test this? I want to just patch the files so I can revert them later, as I use Octoprint for all my printing and wouldn't want to be running HEAD, for obvious reasons.

foosel, 2017-02-20:
You should just check out the `maintenance` branch, install it and restart your server (should display something like `Version: 1.3.2.dev45+g9f96369 (maintenance branch)` then, or a higher number and different commit hash after the `dev`), then when you are done testing check out `master` again, install it and restart the server. If it says "1.3.1" again it's exactly what you had before. For reference: `master` is ONLY updated on new releases (or hotfixes, that has only happened once so far), `maintenance` is what is going to be the next stable release and only gets fixes and improvements. `devel` is what's actually bleeding edge.

Since there can always be some weird kind of merge or code conflict when manually applying commits on a different base I'd prefer you test that way since that would actually verify that what is going to be the next stable version 1.3.2 solves your problem (which is the goal here), not that some "Frankenstein version" that will never actually exist like that does or doesn't solve it.

But if you insist: Related commits are the two cross-referenced above plus 7f0eb8d89560ca0dfa7e511689e315fe0505ea7d for escaping unprintable characters in the terminal tab for better debugging. If you apply those manually and it works, yay, if you do apply those and it doesn't I guess we'll have to wait until 1.3.2 is released to verify if you don't trust `maintenance` for whatever reason.

skorokithakis, 2017-02-20:
My problem is that I use Octopi, which does some weird things with packaging to get the latest version, and I don't want to mess that up. I'm just going to run the standalone server now, though, it should work fine. Thank you!

foosel, 2017-02-20:
On OctoPi the only weird thing is the virtual environment maybe, but other than that it's straight forward:

```
source ~/oprint/bin/activate
cd ~/OctoPrint
git checkout maintenance
python setup.py clean
python setup.py install
sudo service octoprint restart
```

Perform tests. Then to get back to stable:

```
source ~/oprint/bin/activate
cd ~/OctoPrint
git checkout master
python setup.py clean
python setup.py install
sudo service octoprint restart
```

I do this pretty much all the time on my local test RPis, jumping around multiple branches like that. If you run into any problems there, that should be investigated.

skorokithakis, 2017-02-20:
I tried three times, and it looks like this is fixed! I used to get the problem every single time before, and I didn't it at all now, so I'm fairly confident it's good, thank you!

foosel, 2017-02-20:
🙌

commit e1eaf4eb (2017-02-20): comm: More resilience against garbage on the line during connect

  * leading and trailing null bytes will now be stripped
  * while connecting, a "wait" will also trigger switch to connected
    state - relevant when connecting without reset and initial "ok" for
    M110 gets mangled
  * extended virtual printer by new options to test the above:
      * mangled "ok" can now be injected via "preparedOks" setting and
        prepare_ok DEBUG command
      * simulated reset lines on beginning ("start" etc) now only sent
        if new setting "simulateReset" is true

Should fix #1770
closed 2017-03-16 by foosel