OctoPrint/OctoPrint #1771: _config_version not being set after on_settings_migrate()

kantlivelong, 2017-02-13:
Not entirely sure if it just my plugin or an issue with OctoPrint as I can't find any plugins that utilize this other than SoftwareUpdate.

I've tried explicitly setting the value within on_settings_migrate() and nothing gets written.

[Source](https://github.com/kantlivelong/OctoPrint-PSUControl/blob/master/octoprint_psucontrol/__init__.py#L366)

kantlivelong, 2017-02-13:
I guess the version is important to know. :)

OctoPrint 1.3.1

kantlivelong, 2017-02-13:
Log: https://paste.debian.net/914167/

foosel, 2017-02-13:
Could you try the maintenance branch? I think I fell over that myself and fixed it there: https://github.com/foosel/OctoPrint/commit/a646fff5ffed49b31518fdf3bea9ee5cb9da4c96

kantlivelong, 2017-02-13:
Yep looks good in maintenance branch. Are there any plans to backport the fix to 1.2.x for users who have not upgraded?

foosel, 2017-02-13:
There's no way to do that since the only way to get fixes to people that doesn't involve very hackish things (and that wouldn't even work here) is to have them update. And to be quite honest, even if it was possible I'd not open that can of worms, it would only add yet another branch to maintain which would pull even more resources away from actual development. It just doesn't scale :(

kantlivelong, 2017-02-13:
Would you have an idea of an acceptable workaround? I tried setting _config_version in on_settings_migrate() and it did not work on 1.3.1 master.

closed 2017-02-13 by kantlivelong