OctoPrint/OctoPrint #1773: Software Update lists no software/plugins if plugin is installed in the ~/.octoprint/plugins directory.

kantlivelong, 2017-02-13:
Having a plugin that uses SoftwareUpdate installed in ~/.octoprint/plugins/ causes an exception on /plugin/softwareupdate/check [GET].

OctoPrint Version: 1.3.1
Log: http://paste.debian.net/914206/

MoonshineSG, 2017-02-14:
plugins installed and checked by plugin manager are packaged under the site-packages. 
Manually installed plugins (in .octoprint/plugins) can't be checked for updates as they don't provide the needed information.

kantlivelong, 2017-02-14:
Correct but having a plugin in ~/.octoprint/plugins shouldnt break update checks for all plugins.

MoonshineSG, 2017-02-14:
check the logs. it's likely your plugin breaks the update check. 

my guess: you have everything prepared to package it, that includes `get_update_information` and your `get_version` is:

```
	def get_version(self):
			return self._plugin_version

```

since the plugin was not setup, it has no version.... try:

```
	def get_version(self):
		if self._plugin_version:
			return self._plugin_version
		else:
			return "99" #manualy installed
```

kantlivelong, 2017-02-14:
But shouldn't OctoPrint catch this exception and ignore it so that it does not break other plugins?

MoonshineSG, 2017-02-14:
maybe... but it's your plugins `get_update_information` that calls the method that crashes. `				current=self.get_version(),` in your update info ?

kantlivelong, 2017-02-14:
Agreed on that point.

foosel, 2017-02-14:
The cause indeed is an invalid piece of update information here, but OctoPrint really should be more resilient about that. Which it now is as of the above commit to `maintenance`. Will be released with 1.3.2 (and also merged to `devel` ASAP)

closed 2017-03-16 by foosel

CodeBeau, 2018-02-25:
Having a similar problem here. Plugins install properly, but I can't view them like I should in the bottom left corner of settings.
[image]

foosel, 2018-02-26:
Your plugin is marked as disabled, to be enabled after a necessary restart. So of course it doesn't show up, it's not active yet.

commit 5d97dc4b (2017-02-14): Better resilience against broken software update configs

Fixes #1773