OctoPrint/OctoPrint #1774: [REQUEST] One GPIO setting to rule them all

MoonshineSG, 2017-02-14:
As more and more plugins are being developed and some use GPIO, there's been a couple of conflicts due to the multiple calls to GPIO.setup() and some GPIO.cleanup(). There's also BCM vs Board pin numbering.

Maybe OctoPrint could have a generic setting (under "features" ?) to setup and allow the user to choose BCM or Board (which ever they are more comfortable with). A new type of plugin "GPIOPlugin" could indicate to OctoPrint that there are some plugins who need the GPIO setup ... 

The GPIOPlugin could set some rules for developers to follow to avoid GPIO conflicts.

Open for discussion... 

see https://github.com/kantlivelong/OctoPrint-PSUControl/issues/5 

@kantlivelong @Kunsi

kantlivelong, 2017-02-14:
I like the idea of a generic built-in GPIO plugin that would act as a wrapper for various GPIO frameworks (RPi.GPIO, Adafruit_BBIO.GPIO, etc..). Not sure of how much effort this would entail.

Additionally/Alternatively OctoPrint could allow plugins to have other plugins as dependencies.

foosel, 2017-02-14:
It's not a full dependency model, but [the helpers model](http://docs.octoprint.org/en/master/plugins/helpers.html) is actually something that could be of use here. Basically having a GPIO plugin interface defined as helper methods, and then plugins that implement that and take care of stuff.

Instead of modelling dependency management into OctoPrint's plugin implementation I think I'd rather go with trying to somehow reuse the existing pip functionality for that and somehow turning the plugin repository into a package index added during plugin installations (because I don't see it happening to get all plugin authors to register their plugins on the official pypi at this point, although long term that would be great). The reason for that is simply that getting dependency management right isn't as trivial as it seems to be on first glance, and not necessary when there already exist dedicated tools for that.

With that approach you could simply specify a requirement like "OctoPrint-RPiGpio" in your plugin's setup.py and pip would take care of the rest (or log an error if your plugin's dependency isn't installable).

As a side note, I've been meaning to push OctoPrint's releases onto pypi for months now so that "pip install OctoPrint" will work, but I still haven't found the calm day I'd need for that ;)

MoonshineSG, 2017-02-14:
Ha... I completely missed the helpers section... yes, i think that can work - as long as everyone using gpio calls the helper to get the user preference :D 

As for the dependency model, that can be the long term plans, for when you're so free that you're bored ...

foosel, 2017-02-14:
> for when you're so free that you're bored 

To be honest I haven't had such a moment in the past four years ;)

MoonshineSG, 2017-02-14:
and that's why we don't have plugin dependency ! :D

MoonshineSG, 2017-02-14:
the helper could:  initialise the GPIO with the BCM/Board user preference, and maybe even return a list of used pins and/or do a quick conflict check of the currently used pins ... 
oh..  version check, platform check...

foosel, 2017-02-14:
But it is why you even have a system in place for which the need for dependency modeling starts to appear :P

MoonshineSG, 2017-02-14:
and an awesome one it is!

kantlivelong, 2017-02-22:
I've committed a solution to the OctoPrint-PSUControl/devel branch that converts BOARD to BCM if the mode has already been set for BCM. It will work if other plugins call setmode() before PSUControl but won't work in the other direction without the same functionality added to other plugins.

[0cfb7af74bff895d6ca93b8cd40263f33b40bf7b](https://github.com/kantlivelong/OctoPrint-PSUControl/commit/0cfb7af74bff895d6ca93b8cd40263f33b40bf7b)