OctoPrint/OctoPrint #1776: Camera control not visible in German translation

f4l0, 2017-02-15:
#### What were you doing?

Camera control is not visible/available in the German translation. Switching user's language to English solves this...

#### What did you expect to happen?

Camera control in German version

#### What happened instead?

No camera control in German translation

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (master branch)

#### Printer model & used firmware incl. version

#### Browser and Version of Browser, Operating System running Browser

Tested with Chrome, Firefox an Win10 and Ubuntu.

#### Link to octoprint.log

Necessary?

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

foosel, 2017-02-15:
Probably a caching issue, try shift reload in your browser. Please report back/close this issue if that solves it.

foosel, 2017-02-15:
And just for the record: *always* include octoprint.log, it's rarely not necessary and including it right away as asked for ensures faster analysis.

f4l0, 2017-02-16:
Clearing the cache helped! Maybe this should be pointed out somewhere in the wiki!

Thanks for the great piece of software!

closed 2017-02-16 by f4l0

foosel, 2017-02-16:
> **wiki**
> *ˈwɪki/*
> noun
> a website or database developed collaboratively by a community of users, allowing any user to add and edit content.

:)