OctoPrint/OctoPrint #1778: Octopi Trouble

jedi5200, 2017-02-18:
Hi!

I've been having a lot of trouble lately. First, I flashed  the stable Jessie image, but I couldn't even use startx (said the bash didn't recognize it), so I flashed the wheezy stable image. I finally got the wifi up after a long time by deleting the mac-made quotation marks around my ssid and psk in octopi-network.txt and replacing them with pi-made quotations. I puttied into my pi and got into the web interface. I tried to update it, but the update failed. my pi restarted for something and I haven't been able to get back to the web interface since. I thought it was the port number that was throwing me off, but I even edited config.yaml and set a port number, but that did nothing. I also haven't been able to use my raspicam v2. 

I have no printer, but one is coming soon and I am trying this in expectation. it will be an anet a8.

Thanks in advance!

jedi5200, 2017-02-18:
oops, didn't realize this wasn't for octopi. where should I post it instead?

jedi5200, 2017-02-18:
moving this to octopi forum.

foosel, 2017-02-20:
As already mentioned, wrong location since this is the Octo*Print* bug tracker. Continued as thread [on the mailinglist](https://groups.google.com/forum/#!topic/octoprint/eZ-gmwofvZo). Cheers!

closed 2017-02-20 by foosel