OctoPrint/OctoPrint #1779: Loading/Printing files from SD with multiple folders will not work

drdelaney, 2017-02-18:
#### What were you doing?

Trying to load or print a gcode file from SD which had more than one folder. IE /folder1/folder2/file.gcode

#### What did you expect to happen?

The file to load or start printing.

#### What happened instead?

The status bar reset, but kept the previous loaded file (or none if booted or cleared).

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (master branch)

#### Printer model & used firmware incl. version

Prusa i3 Mk2 Clone, RAMPS 1.4 baord, Marlin 1.0.2 firmware.

#### Browser and Version of Browser, Operating System running Browser

Chrome 56.0.2924.87 (64-bit)
Firefox 47.0.1
Internet Explorer 11.576.14393.0

#### Link to octoprint.log

https://gist.github.com/drdelaney/ebb8236a6ada58329d21c6b99d30211a

NOTE: It is not showing the attempt to load the file in here.

#### Link to contents of terminal tab or serial.log

https://gist.github.com/drdelaney/6cb946e19e51381956c8d4048714c93a

#### Link to contents of Javascript console in the browser

Do not know how to obtain this in chrome

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2017-02-20:
That's an issue with your firmware, not OctoPrint:

```
Send: M23 /fidget/done/32m-mg~1.gco
Recv: echo:Now fresh file: /fidget/done/32m-mg~1.gco
Recv: fidget
Recv: done
Recv: open failed, File: done.
Recv: ok
Send: M105
```

OctoPrint tells the firmware "please open /fidget/done/32m-mg~1.gco". Your firmware appears to successfully find `fidget` but then falls on its face when it attempts to open the `done` sub folder in that folder, throws an error and gives up on that `M23`.

Nothing I can do from OctoPrint to solve this, if the firmware won't open the file it won't open the file.

drdelaney, 2017-02-20:
Thank you for the update.  I will go try a firmware update and see what happens.

closed 2017-02-20 by drdelaney