OctoPrint/OctoPrint #1780: Video Stream gets added each time you navigate to the Control tab

ghulands, 2017-02-18:
#### What were you doing?

Everytime you navigate to the Control tab, it adds a new stream to the web page. This can be see in the web inspector and also in the bandwidth on the network interface - jumping about 2MB/s for each stream.

#### What did you expect to happen?
Only use the single stream.

#### What happened instead?
See above.

#### Branch & Commit or Version of OctoPrint
1.3.1

#### Printer model & used firmware incl. version
N/A

#### Browser and Version of Browser, Operating System running Browser
Safari 10.0.3, macOS Sierra 10.12.3

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

ntoff, 2017-02-19:
works normally in Chrome on windows. Navigating to the control page opens up the stream, navigating away from the control tab closes the stream after a few seconds. Navigating to it, away from it, to it, away from it in quick succession doesn't open up any new streams.

OssenFoss, 2017-02-20:
I have noticed the same thing. I tried Chrome (Version 56.0.2924.87 (64-bit)) and Firefox (51.0.1 (64-bit), both work fine. It only affects Safari (Version 10.0.3 (11602.4.8.0.1))

ghulands, 2017-02-20:
Just another thing to add...

It will even duplicate the stream if you leave the control tab selected, but navigate away and back with the safari tabs.

foosel, 2017-02-20:
Looks like a Webkit bug. [First report I found](https://lists.apple.com/archives/web-dev/2006/Oct/msg00138.html) is from 2006. And [here's another ticket](https://bugs.chromium.org/p/chromium/issues/detail?id=73395) about the same bug in Chromium, since fixed. Apparently no fix for Safari.

Thing is, OctoPrint tries to unload the mjpg stream when a) the control tab is not visible or b) the whole browser tab containing OctoPrint is not active for at least 5s. Reason for that is that it doesn't make sense to keep streaming if nobody is watching. Unloading is done by setting the img src to an empty string (and yes, that is the recommended way to do that). That works fine in all browsers - well, besides Safari apparently, because contrary to everything else on the market Safari simply keeps on loading the stream. On return to control tab/browser tab, OctoPrint will set the img src attribute back to the stream's URL. Apparently THAT is something that Safari in fact DOES react to and it loads a second (and then a third and a fourth and so on) stream.

So... it's a big fat browser bug and the only way around it that I see right now is explicitly switching the mjpg unload behaviour off for Safari until this bug is fixed in that browser.

ghulands, 2017-02-20:
I just submitted the PR.

foosel, 2017-02-21:
PR is merged on both `devel` and `maintenance` and this workaround will be in 1.3.2

Thanks! :+1:

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel

derfred33, 2017-05-30:
How about tricking safari to open a dead/non existent stream when the control tab is not on focus or when another tab is clicked and again loading the correct stream when the control tab is clicked?

This way it would not cause load when the tab is not active and would load when the tab is active

I don´t know if this will wrk but its just an idea.

foosel, 2017-05-30:
@derfred33 Problem is that Safari simply continues streaming the stream, even if you set its URL to an existing other image :/ I just tested this on BrowserStack. So apparently the only option for now is to not touch things again once the stream is running in that particular browser. I suggest to use a browser that doesn't have this bug until Apple bothers to fix it (whenever that will happen, considering it's been present for more than ten years now). Chrome and Firefox both don't have any issues.

derfred33, 2017-05-31:
I have this issue on both osx and ios ipad but its not what I often see reported from years ago. They usually say that the stream stops after a few seconds, my stream does not stops. Only if I change the browser tab it stops, then I have to reload the octoprint tab.

This is why I was suggesting a javascript solution to reload the div where the img stream is when the Control tab is clicked.

Something like this:
https://stackoverflow.com/questions/12407368/refresh-div-with-javascript-button

I´ll try to do this as soon as I have a spare pi, next week or so.