OctoPrint/OctoPrint #1781: [Request] ssh from web face

userosos, 2017-02-19:
Can do ssh access to raspi from web-face? It will be very good.

foosel, 2017-02-20:
It will indeed be a very good way to get hacked. Way too insecure. Not going to add this. If anyone wants this, there's a plugin system, but I'll not merge a plugin like that on the repository either. It simply is way way way too risky considering how many people do port forwards without even a remote consideration for security, I don't want to imagine what would happen if those same people had a full featured SSH client in their OctoPrint instances.

Additionally, OctoPrint is cross platform, it does not exclusively target Raspberry Pis running Linux, and something like e.g. a Windows machine wouldn't even have an SSH server to connect to, which makes this a case of "plugin only" anyhow.

closed 2017-02-20 by foosel

userosos, 2017-02-20:
But can do ssh authorization and it will have security.

foosel, 2017-02-20:
No, it won't. Unless you are using a browser extension (in which case, just use that directly) there's no way that I'm aware of that you can open an actual fully secured SSH session from just JavaScript and HTML5. Which means you'll need a server on the other end that acts as your actual SSH client. Which means you'll need to transfer anything you type into your "SSH" window to that server first. Which means it will be plain text unless you put HTTPS in front of your OctoPrint instance with a proper certificate which you CHECK. Unless you store that data on the server side. In which case it won't even be necessary to capture the stream.

Embedding an SSH client into something like OctoPrint would only make sense if there was a way to get an actual end-to-end SSH session going between the web application and the SSH server in question. But that's not possible due to the restrictions of the browser sandbox, at least as far as I'm aware of. All existing solutions appear to depend on a server side component like described above or something like a browser extension.

I can't keep people from putting their unsecured OctoPrint instances on the net. I can't keep them from forwarding port 80 and not 443 (HTTPS) on their OctoPi instances. But I don't have to make it even easier for anyone trying to exploit those people by including a way to gain full system access with a convenient shell.