OctoPrint/OctoPrint #1783: [brainstorming]Install zipped plugin via drag and drop upload?

ntoff, 2017-02-19:
Testing a plugin modification and I'm sitting here thinking "how cool would it be if I could just drag this zip file onto octoprint to have it auto install instead of having to go into the settings and install from zip"

Would that be too dangerous? Too hard to implement? Drag a carefully named zip file like "OctoPrint-PluginName.zip" onto the window and have it auto detect (from the name) that it's a plugin and to install it? Or just assume that "hey this is a zip file, let's just try install it"? Or would this only be something very very few people would even want or need?

Just another crazy sleep deprived idea.

foosel, 2017-02-20:
It would be very tricky if not outright possible in the current implementation since it's currently not possible to intercept/override the globally registered drag-n-drop handler bound to gcode file uploads from within a plugin (which the plugin manager is).

Additionally I think (I might be mistaken though) that only a small handful of people with very special interests would actually ever use that.

Personally what I do in such cases (when NOT testing with a local OctoPrint instance, because for local I just `pip install -e .` from the plugin folder into the instance venv) is `python setup.py sdist` in the plugin folder and then install the resulting archive through the plugin manager. The nice thing about at least Chrome's file selector is that it remembers the last folder, so as soon as I've done that once, for any follow-up installs the navigate-to-file part also is not necessary any more.

ntoff, 2017-02-21:
yeah the trouble is I had to test it against 2 different bits of hardware / software. Usually I just mess about with it on windows and use the develop option, but I needed to test the SoC temperature reading getterer thingy from that navbar temperature plugin to make it work on my AllWinner A20 board running armbian so I had to modify the code, then upload it to both the A20 board and the raspberry pi to make sure I hadn't broken the raspberry pi functionality. I think next time I'll set up a network share and just point the install to that, which would also solve the issue I had of the plugin not showing up after I broke it :)

closed 2017-02-22 by ntoff