OctoPrint/OctoPrint #1784: Stream control from webinterface - cant start, can stop

MikeDabrowski, 2017-02-19:
#### What were you doing?

I just upgraded to 1.3.1 octoprint, got ps camera or prestigio something (doesn't matter rly). They both work fine when i start stream from terminal. But I added start / stop buttons as described in wiki and start button does not work.

#### I tried 
I edited webcamDeamon function runMjpgStreamer:
`
function runMjpgStreamer {
    #input=$1
    #pushd $MJPGSTREAMER_HOME
    #echo Running ./mjpg_streamer -o "output_http.so -w ./www" -i "$input"
    /home/pi/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -i "/home/pi/mjpg-streamer/mjpg-streamer-experimental/input_uvc.so" -o "/home/pi/mjpg-streamer/mjpg-streamer-experimental/output_http.so -w ./www"   
    #popd
}
`
It didn't help. Exactly this line turns stream on when ran from terminal.

#### What did you expect to happen?
I expected it to start stream

#### What happened instead?
Got confirmation of successfully starting stream but /webcam/action?=stream was not working.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (master branch)

#### Browser and Version of Browser, Operating System running Browser
Firefox 53.0a2 (2017-02-18) (32-bit) on Windows 10

#### Link to octoprint.log

http://pastebin.com/tAJrCWgr

#### Link to contents of Javascript console in the browser

Nothing there

foosel, 2017-02-20:
Could you please provide a link to the wiki node you are referring to here? Sounds like it needs to be updated/fixed.

> I edited webcamDeamon function runMjpgStreamer

That should not be necessary and in fact I strongly recommend against editing that file at least on OctoPi since you'll disrupt out-of-the-box system operation.

Are you running OctoPi? A simple `sudo service webcamd start` and `sudo service webcamd stop` should work perfectly fine as commands for the system menu so start/stop your webcam.

MikeDabrowski, 2017-02-20:
Here's the link: 
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#webcam

I am using rpi zero and installed software manually then upgraded via web interface.
I want to point out that stream works if I start it from terminal with command. I just want to use only webinterface to do that.

ntoff, 2017-02-21:
>Could you please provide a link to the wiki node you are referring to here? Sounds like it needs to be updated/fixed.

I manually installed octoprint on armbian the other day using the raspbian instructions (plus a few extra steps) and that mjpeg streamer script straight from the wiki works fine here. I click the system menu -> start stream and the stream starts up as it should.

I also have a raspbian manual install buuuuut I wouldn't compare that, I thieved the service goodies out of the octopi version for that.

@MikeDabrowski how did you add your controls? Did you manually edit config.yaml or use one of the editor plugins? It's pretty easy to mess up config.yaml if you're manually editing it, and especially easy as I found if your editor of choice uses weird quote marks or line endings.

MikeDabrowski, 2017-02-21:
In the wiki (link I gave) there is a section starting with :
`If you want to be able to start and stop mjpeg-streamer from within OctoPrint, put the following in /home/pi/scripts/webcam:`
That is what I did. There is part about configuring config.yaml. I edited it in sublime text 3

ntoff, 2017-02-21:
Yes, that home/pi/scripts/webcam is just the *script*, I was asking how you added the *controls*, i.e. the "button" as you called it in the OP because I actually tried custom buttons and they don't seem to work, it only works as a system menu entry so I was just curious as to whether you'd added them to the system menu entry which appears up in the navbar or whether you'd tried to make them buttons which appear on the "control" tab.

MikeDabrowski, 2017-02-21:
Just scroll down there. There is part about configuring yaml.

foosel, 2017-02-21:
> I manually installed octoprint on armbian the other day using the raspbian instructions (plus a few extra steps) and that mjpeg streamer script straight from the wiki works fine here. I click the system menu -> start stream and the stream starts up as it should.

Perfect, thanks @ntoff. I wasn't sure if there were maybe some other (misleading) infos somewhere on the wiki because that line by @MikeDabrowski 

> I edited webcamDeamon function runMjpgStreamer:

sounded like nothing I'd put into that wiki node and since it's a wiki it's always possible that someone added wrong info or that something changed. But looking at that wiki node in question it actually does look like it should work fine that way on a manual install.

According to the `octoprint.log` provided up there the command is also executed just fine. So right now I have no idea why it's not actually doing anything IF the exact same command called exactly as stated in `config.yaml` works fine from the command line (which should also indicate that the `chmod` commands were successfully executed).

MikeDabrowski, 2017-02-21:
May it be something wrong with the paths? I could not start the stream when not in directory shown in the wiki (experimental).

foosel, 2017-02-21:
> May it be something wrong with the paths? I could not start the stream when not in directory shown in the wiki (experimental).

Yes, which is why I said you should not need to change anything in the `webcamDaemon` script as that might actually break things.

So let's start fresh to rule out any modifications that are interfering here (because otherwise it becomes absolutely impossible to understand what is happening on your end). First make absolutely sure please that your files look _exactly_ like on the wiki. No adjustments. Best create them again. Then make sure the path configured as `MJPGSTREAMER_HOME` in `/home/pi/scripts/webcamDaemon` matches the location where you put your mjpg-streamer (should be `/home/pi/mjpg-streamer/mjpg-streamer-experimental` if you followed the guide, but let's make sure). Make sure both `/home/pi/scripts/webcam` and `/home/pi/scripts/webcamDaemon` are executable by running the `chmod` commands mentioned in the guide.

Plug your camera in, stay in your home directory (`cd`), test `/home/pi/scripts/webcam start` and `/home/pi/scripts/webcam stop` does what it's supposed to do. If it doesn't, stop and report back what it does instead, provide whatever it prints to the screen.

If it works then, try the system commands in OctoPrint again. Report back please.

MikeDabrowski, 2017-02-21:
Ok so :
1. Purged everything
2. installed fresh octopi
3. updated it 
4. ran first command from instruction ` sudo apt-get install subversion libjpeg8-dev imagemagick libav-tools cmake `
It said all was installed so I did ls -al and mjpg-streamer dir was already there. So I went into it and ran `./mjpg_streamer -i "./input_uvc.so -y" -o "./output_http.so" ` with 'y' because I'm using ps3eye . Didn't work:
`
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Device or resource busy
Unable to set format: 1448695129 res: 640x480
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
`
5. Moved mjpg-streamer to another dir and cloned from git. Ran this:
`cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make`
6. stream starts wit hcommand ./mjpg_streamer -i "./input_uvc.so -y" -o "./output_http.so" 
7. created webcam and webcamDeamon scripts. Purely copy paste, nothing changed. 
8. Chmod +x both files. At this point I noticed my mistake: I made script called 'webcamDeamon' not 'webcamDaemon'. Fixed that.
9. Edited config.yaml 
`
system:
  actions: 
   - action: streamon
       command: /home/pi/scripts/webcam start
       confirm: false
       name: Start video stream
   - action: streamoff
       command: /home/pi/scripts/webcam stop
       confirm: false
       name: Stop video stream
`
10 restarted device
11 sudo reboot once more. Cant open octoprint webinterface now ;/
12 removed config.yaml modifications and it started just fine...

Yes, now got problem editing config.yaml 
`
system:
  actions: []
`
Last time I edited it before upgrading and it looked very different. I don't know what to do with square braces.

MikeDabrowski, 2017-02-21:
Edit: Looked through older config.yaml and thought maybe indentation matters. Removed few spaces and saved it as this:
```
system:
  actions: 
  - action: streamon
    command: /home/pi/scripts/webcam start
    confirm: false
    name: Start video stream
  - action: streamoff
    command: /home/pi/scripts/webcam stop
    confirm: false
    name: Stop video stream
```

13 it worked
14 stream is not working when started with the button

MikeDabrowski, 2017-02-21:
So summing up:

The problem was entirely on my part:
 - Typo in webcamDaemon script's name
 - webcamDaemon script was not executable

Thank you all for you guidance.

closed 2017-02-21 by MikeDabrowski

ntoff, 2017-02-22:
@foosel You mentioned the wiki being updated, should it maybe include the same kind of daemon that octopi uses? https://github.com/ntoff/mjpeg-streamer-octoprint-daemon A mere line change here, a path alteration there and your daemon thingy (well I assume it's yours) appears to work under raspbian with a manual octoprint install.

foosel, 2017-02-24:
@ntoff good idea - want do take care of that? :)

I'm starting to wonder though if it wouldn't maybe make sense to split up that "Setup up on Raspbian" page into sub pages, stuff starts to get way too long and intimidating. I mean, the actual OctoPrint setup is only the very first bunch of lines, all the rest is just additional (optional) environment around it. Gets hard to parse if you don't need all of that.

ntoff, 2017-02-24:
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#system-commands

Do the system commands entries still need to be in the wiki? They're handled automatically now aren't they?

foosel, 2017-02-28:
@ntoff about the system commands, we no longer need them as actual system commands, but the commands need to be configured. Can be either done through Settings > Server or through `config.yaml`. I saw in your version you already mention that :+1: - I think the Settings routine is fine and less error prone but it should be specified what commands need to be set.