OctoPrint/OctoPrint #1787: Octoprint resets Skynet Anet A8/Marlin reprap and does not connect

disco22522, 2017-02-21:
#### What were you doing?
Connected newly bought printer with flashed board to Skynet v2.3, setup printer profiles, hit connect. Printer resets. I have timeout timed for longer than usual, as well as "Wait for start on connect". 
[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?
Octoprint to be able to connect to printer Anet A8
#### What happened instead?
Anet A8 reset after hitting "connect".
#### Branch & Commit or Version of OctoPrint
Most recent
[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version
Anet A8 Skynet v2.3.1

foosel, 2017-02-21:
That is sadly a pretty normal behaviour. A lot of printer controllers out there are intentionally built such that they reset/reboot when a serial connection is established. There is nothing I can do from OctoPrint to prevent this from happening. Some printer controllers do have a jumper or similar to disable this behaviour, maybe your's does too. But if it doesn't, well, nothing to be done about this.

foosel, 2017-02-21:
Ah... I missed the "does not connect part", sorry for that. In that case please provide a `serial.log`. 

Also, your `octoprint.log` indicates you have the GPX plugin installed - if your printer runs Marlin firmware (I have NO idea what Skynet actually is) you don't need that and it might actually interfere with regular operations and cause the "unknown symbols". @markwal might be able to comment more on this than me though.

Also:

> **Branch & Commit or Version of OctoPrint**
>
> Most recent

Please in the future just copy and paste the version number. A lot of people think they are on "most recent" and then it turns out they in fact aren't.

Salandora, 2017-02-21:
Skynet is a Marlin fork.
I myself have a Anet A8 and te same problem with stock Marlin. I believe it is the cheap ch340 serial to usb chip that is causing this.

No idea what baudrate skynet is using but for me 115200 seems to make less problems then e.g. 250000.

So maybe try to lower your baudrate.

disco22522, 2017-02-21:
I have GPX plugin installed but disabled due to it being hooked up to a Flash Forge Creator Pro.

The issue was actually that I did not attempt to use a 115200 Baudrate, that's the one I skipped! Silly me, because that was the issue. Thank you for the quick responses, I've been a lurker and love OctoPrint, and love what you guys have done!

Tannoo, 2017-02-22:
I know, I know... This is not an official report, but I have the same issue running 1.3.1 and Marlin's latest RCBugFix. I lowered the baud rate to 115200 (had to change it in Marlin also), and still the same.
Cable is not bad, as I can print just fine directly from the PC using the same cable and at the same 250000 baudrate.

Octopi has been a print saver for me over the last year. This is disheartening.

disco22522, 2017-02-22:
@Tannoo What I would do is just wipe your Rpi and reinstall Octoprint.

Tannoo, 2017-02-23:
@disco22522, acually, it was found that a single change I made in the Marlin firmware of my printer was the cause. I had changed the serial output "start" to "Start" and Octoprint didn't like that. Is was looking for "start" and never found it.

I changed it back and all is good.

foosel, 2017-02-28:
Closing since original issue reported as solved by selecting the correct baud rate.

closed 2017-02-28 by foosel