OctoPrint/OctoPrint #1788: Connection Timeout

Tannoo, 2017-02-22:
#### What were you doing?

I am try to connect to the printer. Terminal shows all the normal status info upon connection to the firmware, even up to getting the SD card status. All looks okay, but 10 secs AFTER all that info is displayed, it errors and never says it's connected.

#### What did you expect to happen?
Connect and continue like it has always done.

#### What happened instead?
```
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
```

#### Branch & Commit or Version of OctoPrint

`Version: 1.3.1 (master branch)`

#### Printer model & used firmware incl. version

FolgerTech Prusa i3, Marlin RC8-RCBugFix

#### Browser and Version of Browser, Operating System running Browser

Microsoft Edge 38.14393.0.0 -- Windows 10

#### Link to octoprint.log

https://gist.github.com/Tannoo/a5a6a6ff9ecf9598fefd217238ecc7b2.js

#### Link to contents of terminal tab or serial.log

https://gist.github.com/Tannoo/865784c84514ad029bd5ff8cf5945489.js

#### Link to contents of Javascript console in the browser

No need for screenshots.

#### Screenshot(s) showing the problem:

The Serial log shows the problem. It's the same I see on the screen.

Salandora, 2017-02-22:
`2017-02-22 07:30:07,501 - Recv: Start`
I guess this is the problem, your firmware sends a capitalized Start instead of a start
@foosel, maybe we should change the 'start in line' to 'start in lower_line' what do you think?

small fix for your links:
**Link to octoprint.log**
https://gist.github.com/Tannoo/a5a6a6ff9ecf9598fefd217238ecc7b2

**Link to contents of terminal tab or serial.log**
https://gist.github.com/Tannoo/865784c84514ad029bd5ff8cf5945489

Tannoo, 2017-02-22:
Oh.... That is something I changed in the firmware of my printer.
If that is the cause, I can put it back.... no big deal.

Tannoo, 2017-02-22:
Right here:

```
  MYSERIAL.begin(BAUDRATE);
  SERIAL_PROTOCOLLNPGM("Start");
  SERIAL_ECHO_START;
```

Tannoo, 2017-02-22:
That was it. Thank you.

closed 2017-02-22 by Tannoo

foosel, 2017-02-22:
As a general rule of thumb: Do *not* modify the messages returned by the firmware on the serial line, or the parser routines within the firmware for messages it receives through serial. And if you do and encounter issues with things like OctoPrint, roll back your changes. You are basically messing around with the only thing resembling a communication protocol that we have here, that can break things and badly. The web wouldn't work correctly either if you changed what your browser sends or how it interprets what it receives.

Tannoo, 2017-02-22:
Yeah. It didn't look to be any protocol thing. It just sent 'start' to the console. I thought I was just making it look better.

I have done MUCH with my firmware. I didn't realize that was an issue until now.

It's a non-issue for me, I just won't change that.

Thank ya'll for ya'lls help!