OctoPrint/OctoPrint #1791: Webcam tries to save/download stream (octet-stream)

JasperAlgra, 2017-02-25:
#### What were you doing?

Watching the webcam stream at http://hostname/webcam/?action=stream in firefox 51.0.1 32bit on windows 10. After a while a download dialog pop-up asking to save the 'octet-stream'. This goes on multiple times (open stream in firefox during the night generates about 25 downloads to be clicked away).

Webcam Microsoft LifeCam studio attached to raspberry pi (octopi). Settings: `camera_usb_options="-r 1280x720 -f 20"`

#### What did you expect to happen?
Not showing a download of type application/octet-stream but just the stream.

#### What happened instead?
It seems sometimes the stream servers gets confused and send the wrong MIME type to firefox so it triggers a download dialog?

#### Branch & Commit or Version of OctoPrint
Version: 1.3.1 (master branch) 

#### Printer model & used firmware incl. version

Ultimaker 2+ (doesn't matter) 

#### Browser and Version of Browser, Operating System running Browser

firefox 51.0.1 32bit on windows 10

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

not applicable

#### Link to contents of Javascript console in the browser

Don't have one ready. Will make one if I capture it again.

#### Screenshot(s) showing the problem:

Don't have one ready. Will make one if I capture it again.

I have read the FAQ.

markwal, 2017-02-26:
That stream url is just a url to mjpg-streamer.  There's no OctoPrint involved.  (mjpg-streamer comes bundled with the octopi image, but that is different than OctoPrint.)  This bug is likely to be a bug in the firefox browser since it's the one that has to decide whether to show the the mjpg stream as an image or a download.  Note: IE always behaves this way since it doesn't support mjpg streams.  But if not a FireFox bug, then it may be a bug in the mjpg-streamer which is here: https://github.com/jacksonliam/mjpg-streamer.

In any case, this can't be a bug in OctoPrint since your steps don't involve running OctoPrint at all.

foosel, 2017-02-26:
Thank you @markwal, you summed it up quite nicely. Therefore closing this, nothing that can be fixed in OctoPrint (and btw also not in OctoPi and probably also not in mjpg-streamer, since it definitely sounds like a hiccuping browser more than anything else).

closed 2017-02-26 by foosel