OctoPrint/OctoPrint #1793: Unable to create a modal window using custom view-model binding

praveenmax, 2017-02-28:
What were you doing?

I am trying to add a custom jinja page with custom viewmodel to create a popup page similar to firstrun.jinja2. It will be opened from navbar menu . As per the docs, we have to push the custom viewmodels into OCTOPRINT_VIEWMODELS. But, it seems that my viewmodel is not getting registered. And so, my modal window is not opening(which may be due to viewmodel not binding).

This is my viewmodel code at **...src\octoprint\static\js\app\viewmodels**
(Code adapted from firstrun.js  )

```
$(function() {
		function InternetViewModel() {
				var self = this;
				
				self.hotspotname = ko.observable("max");
				self.password = ko.observable("123");
		
		}
		
		self.showDialog = function(){
			$("#internet_dialog").modal("show");
		};
		
		self.closeDialog = function(){
			$("#internet_dialog").modal("hide");
		}
		
        self.onAllBound = function(allViewModels) {
                //self.showDialog();
            
        }		
		
		OCTOPRINT_VIEWMODELS.push([
        InternetViewModel,
        [],
        "#internet_dialog"
    ]);
});

```

This is my navbar jinja2 code at the path **...src\octoprint\templates\navbar**
(Code adapted from navbar\settings.jinja2)

```
<a id="navbar_show_internet" class="pull-left" href="#internet_dialog" data-bind="click: function() { $root.show(); }" >
    {{ _('Internet') }}
</a>

```

This is my modal window code at the path **...src\octoprint\templates\dialogs.**
( Code adapted from dialogs\firstrun.jinja2)

```
<div id="internet_dialog" class="modal hide fade" tabindex="-1" role="dialog"  data-backdrop="static" >
   
      <div class="modal-body firstrun">
			<p>Hotspot Name: <input type="text" data-bind="value: hotspotname" /> </p>
			<p>Password: <input type="password" data-bind="value: password" /> </p>
      </div>
	  
</div>

````

What did you expect to happen?

My new jinja2 page to be opened in a modal window.

What happened instead?

Nothing happened on clicking my page button on navbar.

Branch & Commit or Version of OctoPrint

Version 1.2.17 (HEAD -> master branch)

Browser and Version of Browser, Operating System running Browser

Firefox

Link to octoprint.log

https://gist.github.com/praveenmax/b094d85b22d437e0dfc8605fca3488ac

I have read the FAQ.

Thanks for the help

foosel, 2017-02-28:
Have you checked for errors in the JS console? You sadly did not include that as part of your ticket (it's asked for for a reason in the template).

My guess is you are trying to bind your navbar entry twice here (e.g. thanks to stating `customBindings: False` in the template config), and that simply won't work. The JS console should give you valuable hints in that regard. Running OctoPrint with `--debug` will also increase the log output on the JS console, in case what is already there doesn't help.

praveenmax, 2017-03-01:
Hi foosel,
           Sorry for not including js console log. Now I have uploaded the js console log with --debug flag  at this link
[https://gist.github.com/praveenmax/1646aab3fddbb8c41395599addc15b4c]

And the octoprint command prompt log with --debug flag ,
[https://gist.github.com/praveenmax/0121c5656f2536726316e8b3b7aa5f69]

In my js-console, I cant find my viewmodel InternetViewModel. It seems I am missing something.

And , I forgot to post my **octoprint\server\views.py** code. I added a line there for showing a button on navbar. It works but I dont know whether it is safe to do so without harming other functionalities.

```
	templates["navbar"]["entries"] = dict(
		internet=dict(template="navbar/internet.jinja2", _div="navbar_internet", data_bind="visible: loginState.isAdmin"),
		settings=dict(template="navbar/settings.jinja2", _div="navbar_settings", styles=["display: none"], data_bind="visible: loginState.isAdmin")
	)
```

Again, thanks for the timely reply.

foosel, 2017-03-01:
Just noticed:

> Branch & Commit or Version of OctoPrint
>
> Version 1.2.17 (HEAD -> master branch)

You need to update, that's old.

Also - you are directly modifying OctoPrint, NOT using the plugin interface as you are supposed to do. Do not do that, you are messing with internals. That also explains your issues because you haven't even registered your custom JS file from the looks of it as an asset, at least your JS console doesn't indicate it even found your view model. All that stuff would be taken care of if you implemented this stuff as a proper plugin instead of going rogue.

Use the existing plugin API as described in the [documentation](http://docs.octoprint.org/en/master/plugins/index.html) - it's there for a reason, would make your life a lot easier here, and would also not force you to stay on an outdated version.

Closing this because it's not an issue with OctoPrint, and if it is it's an issue with a modified version based on something from around early November - this bug tracker is for issues with supported versions of OctoPrint only, sorry.

If you need further support in any way, the [mailing list](https://groups.google.com/group/octoprint), the [G+ community](https://plus.google.com/communities/102771308349328485741) and - for development questions - the IRC channel (#octoprint on freenode) are much better suited.

closed 2017-03-01 by foosel

praveenmax, 2017-03-07:
Hi foosel,
    Thanks for the detailed reply. I have tried the plugin method but it was causing some error in windows. Will try in my linux machine.