OctoPrint/OctoPrint #1794: Controlling a relay board from your RPi

PowerShellCrack, 2017-02-28:
One thing to be noted in the documentation for "Controlling a relay board from your RPi", is a 10amp relay is not enough amperage for the printer while printing with heater bed turned on; it may pull over 11 amps as it did with mine and it blew my relay. The best suggestion would be to control the board only with this relay and have another PSU for the extruder and/or heat bed.

ntoff, 2017-03-01:
The guide is worded a little weird but it's meant for switching the mains side and if you have over 11 amps on the mains side of your power supply that's 1320 watts at 120volts or 2640 watts at 240 volts, yikes.

From the guide:
>Note: You are playing around with mains voltage here. Errors can be fatal, so if you do not know exactly (!) what you are doing, you shouldn't be doing it!

Though later on it does say to cut the wires coming *from* the power supply, it should read cut the wire going *into* the power supply as the guide is telling you to cut the mains power before it comes in to the power supply, as defined later on:
>As an aside I would suggest to cut the live cable (normally Brown in the EU/UK) and switch that one.

PowerShellCrack, 2017-03-01:
I cut the hot coming from the PSU to the Adurino board. In the instructions it does state "Cut one of the two wires coming from your printers power supply", I figured this was wires on the Audrino board coming from the PSU"; AC side has three wires for main AC powers, and two for DC. I did see the comment below it about the brown wire, but I took it as a suggestion not a requirement. Its also states at the end. "You should be connecting the "hot" side of the power supply - so connect the +V to one screw terminal,.." Based on the PSU labels, that's side is the DC side not the AC side. The main issue with this is I would have to run the RPi on a separate powersupply and I would be running an additional 12 inch of wire to and from the other side of the printer (Relay is mounted where the main board is at. so it could be switched by the rpi (mounted the same side of relay). Now if I were to separate the power, run one V+ power to the Audrino board, then run an additional V+ power wire to a MOSFET, then only the relay switch would use the 5A the board may pull; while the hotbed is using a 11A pull on its own power line and then the last V+ power could power the RPi (with a step-down voltage module of course). Would this work?

[image]

nophead, 2017-03-01:
Yes that will work but the amount of power you would save by turning the 12V rail off would be miniscule if all the motors, heaters and fans are already switched off. The only thing the 12V would then be powering is the 5V rail for the Atmega and the USB chip. If they can also be powered from USB you would in fact save no power at all as the 5V rail would then be provided by the RPI.

PowerShellCrack, 2017-03-01:
I get what your saying. So if I were to isolate the RPi and the relay from the printer, print a case like this: http://www.instructables.com/id/Dual-3D-OctoPrint-Server-With-Powerswitch/?ALLSTEPS. I do have two printer and i was gonna do the same for each. Buy isolating them as a central point, I could save power and control.

nophead, 2017-03-01:
Yes because a big PSU driving a small load like just an RPI is not very efficient. Switching it off and using a smaller PSU to power the RPI should be more efficient. And a separate supply for the RPI makes it more likely it will be able to turn the whole printer off in the event of a fault.

I prefer to use an ATX PSU and power the RPI from the purple 5V standby rail and use it to switch the 12V on an off with just a signal MOSFET on the green PS_ON signal. The 5V standby is actually from a second independent convertor in an ATX PSU. It avoids messing with any mains wiring but you do need dummy loads on the 5V and 3.3V rails to get 12V.

Another solution I like is to use an XBOX PSU as they also have a standby 5V and a control signal, but it is the opposite logic to PS_ON.

foosel, 2017-03-01:
Just for record, if anything can be improved on the mentioned *wiki* article, feel free to do so - it is a wiki, it is editable by anyone with a github account. That [particular page](https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi) is also an actual community effort, the only thing I did was add the big fat "do not play with mains if you don't know the risks or you'll die" disclaimer already mentioned ;)

So - if something's wrong on the wiki, feel free to adjust it yourself :)

I've just done a couple of edits though about the stuff that was mentioned above (and a warning to check the rating of the relay).

foosel, 2017-03-01:
Ah - closed because no TODOs anymore, but feel free to continue your discussion.

closed 2017-03-01 by foosel

PowerShellCrack, 2017-03-01:
so even if the RPi was on its own powersupply and it still switched the 3d printers power on the DC side, is there really that much of kilowatts used on the printer PSU just being powered on? I'm sure there is some, but enough to see a difference in power savings? 

Another idea:
Buy a wireless smart outlet that is controlled by the raspberry pi or hard wire the relay to a house outlet and connect the pi to it.

Source: https://www.samkear.com/hardware/control-power-outlets-wirelessly-raspberry-pi
Source: https://blog.adafruit.com/2016/01/22/home-automation-smart-outlet-control-with-raspberry-pi-piday-raspberrypi-raspberry_pi/

ntoff, 2017-03-02:
It depends what's turned on. Power supplies have stuff inside them that uses power, just to convert the AC into DC. Some people also have lights, PSU fans, it depends on your needs. Get yourself a little plug in power meter and see how much it uses. One or two watts might not seem like much but add it up over a year and if power in your country is expensive, it might add up.

Also technically there's really only 2 wires on AC as well, hot and cold, live and neutral, or however you want to refer to them, the 3rd wire is protective earth an doesn't need to be there in order for the PSU to operate but is there to keep you safe in the event of a fault. Earth (or sometimes called ground) *should* be connected on the power supply but NEVER disconnected via relay or any other means.