OctoPrint/OctoPrint #1797: [request]  a real correct calculation including jerk and acceleration

TylonHH, 2017-03-03:
Hi,
because I love to have the control and a overview of an print I realy like your project. So I wanna give you a suggestion to boost the calculation precision. Maybe it's useful for you too.

Here is actual what I'm using.From this project: https://github.com/davr/Marlin/tree/estimate/Marlin I use an estimate time calculator mentioned and explained [here](http://www.3d-druck-community.de/thread-15074-post-93461.html)(german)

It hits the time on the minute. So maybe you can use this calcutaion for this (or an new) plugin too.
Maybe you can have a look.
If you need information how I use this, let me know.cheers

foosel, 2017-03-06:
There is sadly a huge problem with this kind of approach, also briefly mentioned in [OctoPrint On Air](https://www.youtube.com/watch?v=nCAMjqysBB0&t=2706s): It working depends entirely on knowing about firmware implementation details. 

E.g. your particular tool mentioned up there actually utilizes the Marlin source tree and Marlin's own planner implementation to be able to calculate the correct values based on the GCODE the acceleration, max feedrate and jerk settings. That means this specific tool will work with this specific version of Marlin. It might not work (as in, still produce wrong estimates) for any other Marlin version, and will most definitely not work for any other firmware (Repetier, Sprinter, Smoothie, RepRap, and/or all their forks out there). It will also only work if the firmware settings are known, which means you either need to tell users where/how to find those values, or alternatively read them from the printer, in which case however a printer connection to the exact printer on which a file is to be printed on needs to be already established when said file is analysed or otherwise it's not possible to retrieve the necessary settings.

So, in a nutshell, yes, this can improve the accuracy of the initial time estimation produced during GCODE analysis a lot - for very specific firmware implementations, and everyone else is left out in the cold. Which means a lot of effort spent on a rather unsatisfying end result for the general user base.

I've been meaning to open up the GCODE analysis for plugins to do something like the above where a more specific approach in fact could work if someone where to invest the time needed for it, but it appears I never so far did open up a feature request for that. So how about this: I adjust this ticket to be said feature request, and once that is closed as implemented let's see what the community comes up with in terms of alternative GCODE analysis implementations. Sounds good?

TylonHH, 2017-03-06:
Thanks allot for the detailed reply.
You're totally right, I was just in my little Marlin world and didn't thought so far.
But your suggest sounds really fair and right. It's better to find and universal tool to calculate.

Best regards

foosel, 2017-03-09:
Leaving it open but changed the request as explained :)

bagobones, 2017-05-17:
I like the idea of plugins for estimation.

I did notice that http://gcode.ws/# has been quite accurate for me. I don't often print exactly the same thing so in the current octoprint implementation I am just getting crazy 2 day initial estimates for prints that really take over 10hrs it seems.

foosel, 2018-05-17:
OctoPrint 1.3.9 will ship with two new hooks [`octoprint.filemanager.analysis.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-filemanager-analysis-factory) and [`octoprint.printer.estimation.factory`](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-printer-estimation-factory) that will allow plugins to override both the initial gcode analysis implementation as well as the live print time estimation. 

That should allow people to experiment with this stuff and find better but possibly more hardware/workflow/setup specific approaches that provide more accurate results than a stock option that has to *always* work can.

Knock yourselves out with this ;)

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel