OctoPrint/OctoPrint #1800: [Request] G0 for jogger renders laser user's encounter easier

arthurwolf, 2017-03-04:
#### What were you doing?

Using Octoprint to control a ( smoothie powered ) lasercutter, usinq the jog keys to position the head

#### What did you expect to happen?

The head to move with the laser staying off

#### What happened instead?

The head moved with the laser turning on, thus cutting the plate in places it wasn't supposed to be.

This is because the jogging is using G1, which in Smoothie means «move with tool on».
This can be easily fixed by instead using G0.

I tried it by modifying the following lines : 

pi@octopi:~ $ ack-grep G1 * | ack-grep command
OctoPrint/src/octoprint/printer/standard.py:256:		self.commands(["G91", "G1 %s%.4f F%d" % (axis.upper(), amount, movement_speed), "G90"])
OctoPrint/src/octoprint/printer/standard.py:277:		self.commands(["G91", "G1 E%s F%d" % (amount, extrusion_speed), "G90"])
OctoPrint/build/lib.linux-armv7l-2.7/octoprint/printer/standard.py:256:		self.commands(["G91", "G1 %s%.4f F%d" % (axis.upper(), amount, movement_speed), "G90"])
OctoPrint/build/lib.linux-armv7l-2.7/octoprint/printer/standard.py:277:		self.commands(["G91", "G1 E%s F%d" % (amount, extrusion_speed), "G90"])
oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/printer/standard.py:256:		self.commands(["G91", "G1 %s%.4f F%d" % (axis.upper(), amount, movement_speed), "G90"])
oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/printer/standard.py:277:		self.commands(["G91", "G1 E%s F%d" % (amount, extrusion_speed), "G90"])

to use G0, and it works great.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.10 (master branch) 
On octopi

#### Printer model & used firmware incl. version

Laser cutter using latest smoothie edqe

#### Link to octoprint.log

http://pastebin.com/HT6z5SCa

I have read the FAQ, a long time ago.

ntoff, 2017-03-05:
That sounds more like something smoothiware should not be doing. There should be separate laser on, laser off, and movement commands. Why would you combine a move command to turn the laser on? Is that even within any kind of gcode standard?

foosel, 2017-03-06:
While 3d printer firmware usually implements `G0` and `G1` exactly the same, I distinctly remember from past discussions around the net that `G0` is supposed to be "go to selected point as fast and possible" and `G1` is supposed to be "go to selected point in a straight line". When someone uses manual jog controls, they'll expect the tool head to move in a straight line, so based on that `G1` would be the correct command to use here. If I were to change that it might break for people who rely on the above (e.g. with adjusted firmware on a Delta that implements `G0` as an actual rapid move taking advantage of the non-cartesian motion system).

So the only thing that could be done here is allowing to user-define the commands used for jogging but sticking to the current defaults. And that is something that is more a :+1: for #154 than its own request actually.

As a side-note, I suspect the reason why `G1` has the laser come on on Smoothie is this note (or rather, its reverse interpretation with regards to `G1`) on the `G0` command in the NIST spec?

> It is expected that cutting will not take place when a G0 command is executing.

Is this a common interpretation in the CNC world? Because for me it sounds like this should be read _only_ as "if tool is on before executing a `G0`, switch it off" and not _also_ as "if tool is off before executing a `G1`, switch it on". But then, it wouldn't be the first time that what is written down in a standard can be interpreted in several ways and one common interpretation becomes the _actual_ standard.

nophead, 2017-03-06:
> something that is more a 👍 for #154

In the CNC world the tool doesn't get switched on an off. The spindle it is
mounted on gets started and stopped with M3 and M5. G1 just moves the axes.

hungerpirat, 2017-03-06:
As a laser guy myself I understand @arthurwolf 's usecase. But I think in OctoPrint the arrow buttons are meant for the usecase "manual and precise control". Additionally I agree with @arthurwolf about G0 switching the laser off - but I am unsure if I would bind an implicit "laser on" to the G1 command.

arthurwolf, 2017-03-06:
So, the NIST standard says nothing about laser cutting. We had to figure something out. We chose the solution by consensus, and ended up with one pretty much nobody was unhappy with.

The way Smoothie implements laser cutting, is that G1 cuts, and G0 doesn't. 
This has the following advantages: 
* It allows you to use CNC milling CAM software to generate laser cutting g-code out of the box without modifying *anything* ( which is awesome )
* It's easier on the parser and results in smoother parsing/execution compared to a M3/M5 solution ( parsing speed is super important for lasercutters )
* It allows for easy implementation of jogging ( as long as the host uses G0 for jogging, which most do )

Smoothie started doing this years ago, and since it's been copied by others and is pretty much the de-facto standard in the Open-Source Laser-Cutting world AFAIK. I really doubt anyone is going back on this, the most popular open laser CAM solutions like Visicut and Laserweb do this too.
The same way the Reprap G-code is the standard for 3D printing, this is the standard for laser-cutting. And it has the advantage of being a 8-word standard that works great.

Because "3D printing only" firmwares implement G1 and G0 as synonyms, I really think it'd be harmelss to switch Octoprint to doing G0 for jogging like other host software ( like Pronterface etc ) does.

I don't think the "G0 on deltas is the shortest route" argument is valid ( that part of the NIST is ignored by pretty much all software since the 80s btw. Nobody cares that one axis finishes before the others if they are going to have to wait for the slower axis anyway, so everybody does G0 and G1 the same : interpolated ). No firmware I know of implements G0 that way on deltas, doing so wouldn't result in gains in speed, and doing so would have very nasty side effects ( you'd want to jog but would move in weird curves that would potentially intersect your bed ). Nobody would do that.

If this gets implemented just as a parameter I'd be fine with it, but I *really* think there is absolutely no downsides to making this the default, and a lot of advantages to doing so.

cp2004, 2021-07-16:
This change was implemented with PR #4098 and will be part of OctoPrint 1.7.0

closed 2022-02-17 by cp2004