OctoPrint/OctoPrint #1801: Custom bounding box not saved

matfur92, 2017-03-04:
Hi, 
I'm on Version 1.3.1 (master branch).

When I save my "Custom bounding box" in printer profile, nothing is saved.
I found the problem in rest service, it doesn't save the properties passed.

Here a piece of curl registered : 
INPUT to service :

--data-binary '{"profile":{"id":"_default","name":"MendelMaxMF","color":"default","model":"MendelMax 1.5 RepRap Printer","volume":{"width":250,"depth":230,"height":125,"formFactor":"rectangular","origin":"lowerleft",**"custom_box":{"x_min":31,"y_min":12,"z_min":0,"x_max":243,"y_max":212,"z_max":125}}**,"heatedBed":true,"extruder":{"count":1,"offsets":[[0,0]],"nozzleDiameter":0.4},"axes":{"x":{"speed":6000,"inverted":false},"y":{"speed":6000,"inverted":false},"z":{"speed":200,"inverted":false},"e":{"speed":300,"inverted":false}}}}' 

OUTPUT of service : 

'{"profile": {"axes": {"e": {"inverted": false, "speed": 300}, "x": {"inverted": false, "speed": 6000}, "y": {"inverted": false, "speed": 6000}, "z": {"inverted": false, "speed": 200}}, "color": "default", "current": true, "default": true, "extruder": {"count": 1, "nozzleDiameter": 0.4, "offsets": [[0.0, 0.0]]}, "heatedBed": true, "id": "_default", "model": "MendelMax 1.5 RepRap Printer", "name": "MendelMaxMF", "resource": "http://octopi.local/api/printerprofiles/_default", "volume": {"custom_box": false , "depth": 230.0, "formFactor": "rectangular", "height": 125.0, "origin": "lowerleft", "width": 250.0}}}'

Thanks a lot in advance

matfur92, 2017-03-04:
----

#### What were you doing?

When I save my "Custom bounding box" in printer profile, nothing is saved.
I found the problem in rest service, it doesn't save the properties passed.

#### What did you expect to happen?

I will have the "Custom bounding box" with my params saved

#### What happened instead?

"Custom bounding box" is disabled

#### Branch & Commit or Version of OctoPrint

Version 1.3.1 (master branch)

#### Printer model & used firmware incl. version

MendelMax 

#### Browser and Version of Browser, Operating System running Browser

Mac OS 10.12.3 - Safari and Chrome last version

#### Link to octoprint.log

http://pastebin.com/cp7cJa7a

#### Link to contents of terminal tab or serial.log

empty serial log

#### Link to contents of Javascript console in the browser

empty

#### Screenshot(s) showing the problem:

empty

foosel, 2017-03-06:
You are sending invalid values there. Your custom bounding box is fully contained within the print volume and therefore completely redundant. The custom bounding box is to be used for defining an area _larger_ than your defined print volume where your print head is safe even though it's outside of the dedicated print volume. It also says this in the editor (emphasis mine):

> If your printer's print head may move slightly **outside the print volume** (e.g. for nozzle cleaning routines) you can define a custom safe bounding box for its movements below.

This looks more like you actually defined your print volume too large to be safe for your print head and you need to fix that instead.

In your case, your print volume is

```
x_min: 0
x_max: 250
y_min: 0
y_max: 230
z_min: 0
z_max: 125
```

and the custom bounding box you defined there is:

```
x_min: 31
x_max: 243
y_min: 12
y_max: 212
z_min: 0
z_max: 125
```

That doesn't make a lot of sense, you are basically defining your print volume to be larger than the safe areas for the print head.

matfur92, 2017-03-06:
Thanks! Sorry for my error!  :(
My idea was to replicate this configuration of RepetierHost:
[image]

With these values what are respectively values in OctoPrint?
`
x : ?
y : ?
z : ?
x_min: ?
x_max: ?
y_min: ?
y_max: ?
z_min: ?
z_max: ?
`

Thanks a lot in advance!

foosel, 2017-03-08:
@matfur92 sorry, I can't say that since I don't know how exactly that stuff does work in Repetier Host. In general though, if you need to define a bounding box that is smaller than the coordinate volume defined in your firmware, you won't be able to do that in OctoPrint.

If your print volume includes coordinates that are unsafe for the head to move to, the print volume is simply not defined properly in the firmware in my opinion, since that's the authority on the coordinates the printer should accept. And it shouldn't except anything that might damage it.

closed 2017-03-08 by foosel