OctoPrint/OctoPrint #1806: Upload to SD greyed out

chevyfried, 2017-03-07:
Monoprice Mini Select printer/Malyan m200

----

#### What were you doing?

Trying to upload to SD, it's greyed out.

#### What did you expect to happen?

Upload the file to SD

#### What happened instead?

Nothing, no pop up, no error

#### Branch & Commit or Version of OctoPrint

1.3.1 master branch

#### Printer model & used firmware incl. version

Monoprice Mini v42.29

#### Browser and Version of Browser, Operating System running Browser

Chrome

#### Link to octoprint.log

http://pastebin.com/aam8J38W

#### Link to contents of terminal tab or serial.log

http://pastebin.com/izvwCBsm

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

http://imgur.com/a/as7Hs

ntoff, 2017-03-07:
In settings -> features, do you have "enable sd support" checked? Are you using manual firmware feature selection and if so  do you have "always assume sd card is present" checked?

chevyfried, 2017-03-07:
Yes those are both checked.

foosel, 2017-03-08:
That particular printer has its own firmware that - while claimed to behave compatible to Marlin - actually doesn't. Especially when it comes to SD file transfers, but also other things like line number handling. There already is a ticket about tackling that: #1762. Users with the same printer reported the option mentioned above to work just fine (you need to reconnect to the printer though after saving the settings or they won't take). 

I suggest to continue discussion/debugging in the existing ticket - this is basically a duplicate.

closed 2017-03-08 by foosel