OctoPrint/OctoPrint #1807: [Brainstorming]Better file management for timelapses

ntoff, 2017-03-07:
A request for a delete confirmation dialog https://github.com/foosel/OctoPrint/issues/1798 prompted me to quickly whack a confirmation dialog box on there but then discussion got to checkboxes and such, I don't really know much about how to achieve that so @Salandora if you're listening, is it possible to easily add timelapse file management to your file manager plugin? I noticed it has the ability to click and select multiple files for deletion  and the two step deletion process might be all the safety net that is required.

As it stands my method for mass timelapse deletion is to log in via SSH and just `rm *.mpg` which probably isn't the most elegant solution but it's far quicker than clicking on each files' delete icon and @foosel brought up the issue of a confirmation dialog box hampering efforts to delete multiple files if it were to pop up for every delete request (which doesn't affect me so I never thought about it). There's also a shortcut method using the "system" menu where one can enter the rm command to clean timelapses without having to log in via ssh, still might not be the best solution.

If the file manager plugin can easily be modified to also be able to mass delete timelapses, then I feel a dialog box confirming any timelapse deletion (from the current "timelapse" tab) would be a little bit less of a nuisance if there were another option for deleting multiple files simultaneously.

foosel, 2017-03-07:
I wouldn't modify the file manager plugin for that (or at least not exclusively rely on it). Would probably also be way more work than adapting the current timelapse file list slightly. 

I think just adding check boxes to the entries and a "delete checked" button would work fine considering in the timelapse file list we do have the space for that (in the uploads file list we don't, so things are way more tricky there). 

And then we could add a confirmation dialog for single deletes (and one for bulk delete) and all bases should be covered.

Same for unrendered timelapses (just in case).

Salandora, 2017-03-08:
I fully agree with foosel, I also think the file manager is the wrong place for this.
but checkboxes doesn't look nice too :-P so maybe one could take the way my file manager did to "check" the timelapse entrys and adapt it to the timelapse tab.

I don't know if I will find the time to do this, so I if someone else e.g. @ntoff want's to try it?

foosel, 2017-03-08:
Huh, I think it looks just fine (non-functional mockup only):

[image]

[image]

[image]

In any case though we'll want @BillyBlaze in on this too - for his opinion, and to make him aware because whatever comes out of this, it will most likely necessitate some adjustments in the TouchUI plugin.

BillyBlaze, 2017-03-15:
Thanks for the heads-up. I see no problems with this feature. Looks great 👍

ntoff, 2017-03-16:
Is the individual delete icon still necessary? Having to tick a box and then press a button seems like enough steps that an accidental click is pretty far fetched, but having that delete icon there might be too easy to hit, especially on a touch screen device big enough to not require touchui (I believe the original issue was "missed clicks and a few drinks" or something. https://github.com/foosel/OctoPrint/issues/1798

foosel, 2017-03-16:
That delete button would have a confirmation dialog. As would the "Delete checked" one. First one with something like "You are about to delete timelapseFoo.mpg", second one with something like "You are about to delete 23 timelapses".

At least that's the current idea I'm playing with.

foosel, 2017-03-22:
So I just pushed this to `maintenance`:

[image]

There's still a bug with the caching of the unrendered timelapses that I think was already present and which I have to iron out, but overall I think this should work, or what do you think?

ntoff, 2017-03-22:
>what do you think?

"omg, why are you wasting my time with a confirmation dialog when I've already had to click to select, then click to delete, what next? a confirmation on the confirmation? are you sure you want to delete these files? are you really sure? _this is your last chance to push the cancellation button_"

(spaceballs reference AND a call back to the estop button discussion :) )

I like it, just wish I could have helped in some way (I feel guilty starting a discussion and then it goes to a point too far for my knowledge of copy / paste programming)

ardenpm, 2017-04-12:
The proposed solution looks fine to me and is probably what I would have suggested. Having accidentally deleted an important timelapse when trying to download it, I am in favour of confirmations but also the convenience of being able to select multiple timelapses to delete in one go.

If confirmations were to be removed then a trash, empty trash pattern would work as well (90's as that is). Either way, some timelapses are hours or days of capturing and can be important for understanding why prints fail as well so losing can be pretty painful.

I think the slickest confirmation would be one that was essentially in place (so there isn't the travel of the mouse from the trash icon to the middle of the screen), however with the space allocated now there isn't really room for it anyway.

foosel, 2017-05-11:
@ardenpm oops, I only saw this now. I was never a fan of the trash pattern to be honest, to this day I perform deletes on windows with shift+delete for the simple reason that I always forgot to empty the trash - I fear the same would happen here :)

And anything slicker saving on mouse travel is rather a topic for a complete UI redesign I fear, as you pointed out yourself, we are pretty much at a limit now with regards to what is possible ;)

Anyhow, closing this since I think all has been said and we do have an implementation now that's about to go out the door as part of 1.3.3rc1 :)

closed 2017-05-11 by foosel