OctoPrint/OctoPrint #1808: RPi Zero W does not connect with onboard wifi, has to use dongle.

simo4295, 2017-03-07:
----

#### What were you doing?
I am trying to connect to octoprint using just the onboard WiFi, however with no success,
used a dongle and it connected perfectly.
When flashing the newest armbian build, i am able to use the onboard WiFi, from the desktop.
Is it not currently implemented or am I doing something wrong?

closed 2017-03-08 by simo4295

maxekman, 2017-03-14:
I'm having the same problem. It could be that Octopi needs to be based on the latest Jessie Lite. More info here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=176147.

foosel, 2017-03-14:
> Also *make sure you are at the right place* - this is the bug tracker of the official version of OctoPrint, not the Raspberry Pi image OctoPi

OctoPi bug tracker is here: https://github.com/guysoft/OctoPi/issues/ and ticket about rpi zero w support is here: https://github.com/guysoft/OctoPi/issues/318#issuecomment-286502562

In a nutshell: latest OctoPi nightlies should work, performance could be better, and this is the wrong bug tracker for issues with OctoPi ;)

maxekman, 2017-03-14:
Thanks for letting us know, did not realise the projects were separate.