OctoPrint/OctoPrint #1810: [Request] PrintProgress% Event

RJ-Make, 2017-03-08:
For mqtt processing.

FHeilmann, 2017-03-08:
I am assuming you are developing a plugin anyway so:

You can use the `ProgressPlugin` Mixin  and implement the `on_print_progress()` function to generate your MQTT update. See here:

http://docs.octoprint.org/en/1.3.1/plugins/mixins.html#progressplugin

foosel, 2017-03-08:
What @FHeilmann said. Also, the [existing MQTT plugin](http://plugins.octoprint.org/plugins/mqtt/) exports a bunch of [helper functions](https://github.com/OctoPrint/OctoPrint-MQTT/blob/master/README.md#helpers) to allow easier sending of MQTT messages for other plugins. More on the general concept of helpers can be found in the [plugin docs](http://docs.octoprint.org/en/master/plugins/helpers.html).

Adding an event for print progress percentage would only make sense for full percentages, otherwise it would produce too much noise on the event bus during prints. It still would produce noise on all connected clients for data they have anyways, so I'm not sure yet if it would be a good idea to add such an event solely for allowing progress reporting for registered command line event handlers considering the `ProgressPlugin` mixin already exists.

RJ-Make, 2017-03-08:
Thanks for your responses. I'm not building a plugin, the mqtt topic will be used to calculate (very roughly) some progress functions in my home automation system.

@foosel 5% (maybe even 10%) resolution would be perfectly fine. So the message only needs to be sent on 5,10,15,20 etc..

FHeilmann, 2017-03-09:
If you're not building a plug-in, how are you planning to subscribe to the progress event? Regardless of how, I'm pretty sure just building a simple, no-frontend plugin that does nothing but implement the ProgressPlugin Mixin, and go

```Python
on_progress(progress):
   mqtt.do_whatever(topic, progress)
```

foosel, 2017-03-09:
> If you're not building a plug-in, how are you planning to subscribe to the progress event?

By using regular [system event handlers](http://docs.octoprint.org/en/master/events/index.html#example) I guess :)

FHeilmann, 2017-03-09:
TIL I learned about System Event handlers :stuck_out_tongue:

RJ-Make, 2017-03-10:
@FHeilmann Can I hit you up to build the plug-in.. 👍 :-)  @foosel hit it right on the money, the mqtt functionality is so well integrated, it's hard not to use it.. The mqtt options is one of the reasons I choose to move forward with OctoPrint. Maybe even one day we'll see some bi-directional topics/events.. :-)

foosel, 2017-03-10:
I'd also happily accept any PRs to just have the existing MQTT plugin also implement the `ProgressPlugin` mixin and push progress updates to configurable topics ;)

FHeilmann, 2017-03-10:
After having a quick look I agree that adding the progress to the mqtt plugin should be straight forward. Unfortunately I'm swamped for the weekend so I'll most like not get around to submitting a pull request until early next week.

RJ-Make, 2017-03-10:
@FHeilmann @foosel You guys Rock!

FHeilmann, 2017-03-12:
@RJ-Make I created a pull request in OctoPrint/OctoPrint-MQTT#9. 
Also you can check out the "publish_progress" branch of my fork to test this.

FHeilmann, 2017-03-14:
@RJ-Make the changes were just merged, update your plugin to the latest version and you should be good to go!

RJ-Make, 2017-03-14:
This is going to sound ridiculous, but how do I update the plugin? I'm seeing version 0.2 currently installed? (Running 1.3.2rc1 OctoPrint)

Also what is the topic name that I should use?

Thanks for all your work @FHeilmann this is really going to help me allot.

FHeilmann, 2017-03-15:
verison 0.2 of the plug-in had no auto-update capabilities, so you'll have to uninstall the plugin and install it again.

As for the topics, check the updated info on the landing page of the github repository: https://github.com/OctoPrint/OctoPrint-MQTT
There is a section about the progress topics.

foosel, 2017-03-15:
Yeah, that missing update functionality is a bummer. Kicked myself a bit when I noticed that yesterday. At least going forward that won't be a problem anymore. Clicking "Reinstall" in the repository in Plugin Manager > "Get more..." should also get you updated.

RJ-Make, 2017-03-15:
Thank You @FHeilmann  and @foosel. I will definitely share how I'm using this and other mqtt (OctoPrint) functions to handle everything from ventilation and filtration modes based on print material, to an elaborate notification system :-)  I'll post it on my YouTube channel when I finally get everything sorted.

Thanks again

LazeMSS, 2021-03-31:
@foosel  this could closed I think.

foosel, 2021-03-31:
Agreed.

closed 2021-03-31 by foosel