OctoPrint/OctoPrint #1811: Timelapse settings "save config" button feedback.

rewolff, 2017-03-08:
Hi, 
Possibly already fixed, but.. I'm running 1.3.0-something-git. (on my "production" 3D printer.. It works, don't touch it....). 

I started a print, but before the bed was up to temperature I decided that I wanted to change the timelapse parameters for this print. So I hit cancel (no harm done), and tried configuring the timelapse... Then it tells me I need to wait 20 minutes before that becomes possible.... Sigh. This should be abort-able just like a 3D print. 

In any case the "save config" button never gave any feedback that anything happened.

ntoff, 2017-03-10:
is this a bug report or a feature request? There's never been any feedback for the save config button, apart from the fact that once saved it removes the check box from "save default" and greys itself out.

rewolff, 2017-03-11:
Maybe my eyes are bad that I'm having trouble seeing the difference between grayed out and not. Maybe it WAS grayed out, maybe not. But I tried pressing the "save" button and nothing happened. Took me a while to figure out.

ntoff, 2017-03-15:
You're right, in the default white theme (I've been staring at a dark theme for the past week, that I've been working on) there isn't much of a difference between it being enabled and not, though there is a difference.

I stuck some text on the page that pops up and notifies you that there's unsaved changes https://github.com/ntoff/OctoPrint/commit/381a0169e64835509ac0e0857f5d15f86766f8f0

@foosel You mentioned a spinner (which I didn't do), I've been spoiled by my i7's speed which saves insatntly. Would a spinner icon stay on the screen long enough to be noticed? Even if octoprint is running on a raspberry pi, I don't imagine the save process would take that long would it?. Perhaps that might come and go so quickly that it would be missed?

rewolff, 2017-03-15:
How about changing the text of the "save" button to say what it does and does not do. 

SAVE, use for the next print 
SAVE, use for all prints in the future
SAVE OK, these settings will be used for the next print
SAVE OK, these settings will be used for all prints in the future.

foosel, 2017-05-11:
To be released with 1.3.3 as depicted above.

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel