OctoPrint/OctoPrint #1812: Tab panel in Settings scrolls to top when clicking tabs on low resolution displays.

kantlivelong, 2017-03-08:
#### What were you doing?

1. Set resolution to 1280x800
2. Open Settings
3. Scroll down and click Announcements

#### What did you expect to happen?
Announcements is displayed and the left menu stays in place.

#### What happened instead?
Announcments is displayed and the left menu scrolls to top.

#### Branch & Commit or Version of OctoPrint

1.3.1 (master)

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

http://paste.debian.net/918773/

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

https://s1.webmshare.com/QYXaJ.webm

I have read the FAQ.

foosel, 2017-03-08:
Urks. Stuff like that can be yucky. I'll take a look.

foosel, 2017-03-09:
Perfect, just merged your PR (also backported it to `maintenance`), will be released with 1.3.2

commit 5d1a0eaa (2017-03-09): Only reset scroll position for settings content. (Fixes #1812)

(cherry picked from commit 5c81833)
PR #1814 (2017-03-09): Only reset scroll position for settings content. (Fixes #1812)
#### What does this PR do and why is it necessary?
Fixes #1812 by only resetting scroll position for `#settings_dialog_content` instead of `.scrollable`

#### How was it tested? How can it be tested by the reviewer?
See #1812 

#### Any background context you want to provide?

#### What are the relevant tickets if any?
#1812 

#### Screenshots (if appropriate)

#### Further notes
commit 5c818334 (2017-03-09): Only reset scroll position for settings content. (Fixes #1812)
closed 2017-03-16 by foosel