OctoPrint/OctoPrint #1816: Issue connecting to printer

gayaths, 2017-03-09:
#### What were you doing?
Connecting to my 3D printer

#### What did you expect to happen?

Successfully connect to the printer

#### What happened instead?

Did not connect to the printer. State changed from "Connecting" to "Offline" after a little while.

#### Branch & Commit or Version of OctoPrint

1.3.1 (master branch) 

#### Printer model & used firmware incl. version

Creatbot DX Plus (http://www.creatbot.com/en/creatbot-dx.html)

#### Browser and Version of Browser, Operating System running Browser

Microsoft Edge 38.14393.0.0, Firefox 51.0.1

#### Link to octoprint.log

http://pastebin.com/ADHHXaJT

#### Link to contents of terminal tab or serial.log

http://pastebin.com/4De6uzSt

#### Additional details
I am running Octoprint on OrangPi PC Plus (ARMBIAN 5.25 stable Ubuntu 16.04.2 LTS 3.4.113-sun8i). I have successfully run numerous other printers with OctoPrint without any issues. The main difference I see between this and other printers is that the baud rate is different (250000 vs 115200) and port string (/dev/ttyACMx vs /dev/ttyUSBx). 

I initially assumed it was an issue with pyserial not working at 250000 baud rate, but the serial.log shows that it successfully sends and receives messages from the printer. 

Also I was able to successfully communicate with the printer using pronsole (pronterface console, on the OrangePi itself) and Matter Control (on Windows machine).

foosel, 2017-03-09:
Your printer doesn't adhere to the protocol:

```
2017-03-09 08:27:06,731 - Send: N0 M110 N0*125
2017-03-09 08:27:07,805 - Recv: CreatBot 3D Printer Start >>>
2017-03-09 08:27:07,817 - Recv: Firmware:5.3.170213
2017-03-09 08:27:07,825 - Recv: Last Updated: Feb 22 2017 13:46:31 | Author: CreatBot LYN
2017-03-09 08:27:07,829 - Recv: Free Memory: 3191
2017-03-09 08:27:07,835 - Recv: echo:V20-2 stored settings retrieved (461 bytes)
2017-03-09 08:27:12,243 - Recv: echo:SD init fail
2017-03-09 08:29:42,270 - There was a timeout while trying to connect to the printer
```

It never replies with an `ok` to the initial `M110` - probably because it wasn't yet ready to receive, the startup line only comes after that. But the problem is that it also doesn't send the initial `start` message that would trigger OctoPrint to perform another attempt at an handshake. Instead it sends that completely non-standard "CreatBot 3D Printer Start >>>" message.

OctoPrint doesn't connect therefore because it can't definitely confirm that it's in fact a 3d printer it is talking to and not maybe a coffee maker or a washing machine - it doesn't behave correctly.

You might want to take this up with your printer vendor - they produced a firmware that doesn't behave protocol conform and that is causing the issues here.

I could put in a workaround for this specific printer to get around this issue, but the problem is that then the next printer vendor does it all differently again and the next workaround needs to be added, and at the end you have a completely unmaintainable mess of workarounds for printers whose vendors don't care about adhering to established standards and just let volunteers/third parties do their job instead.

So, scream at your printer vendor, they sold you a somewhat faulty product. Chances are high though they won't care because "they don't support other software", so in the meantime you might be able to connect by placing [this `fix_creatbot_stupid.py`](https://gist.github.com/foosel/76da8da0fb64d7ac74408ff85f1675fa) into `~/.octoprint/plugins` of the user your OctoPrint installation is running under and restarting. No guarantees though - if they broke protocol on the very first line they sent out, who knows what else they broke.

closed 2017-03-09 by foosel

gayaths, 2017-03-09:
Thanks a lot for your response. I will do the fix you mentioned. In the meantime I was able to successfully connect using the workaround mentioned in Issue #1770 (https://github.com/foosel/OctoPrint/issues/1770)
I have also contacted the manufacturer, they haven't got back to me yet, lets see what they will do.
I will get back once I place the fix_creatbot_stupid.py :D in the plugins folder.

gayaths, 2017-03-10:
Hi,
I put the python file you sent in to the plugins folder and now it connects without any issues. I would like to know whether the hook you used read every single line returned from the machine and if it is so, will there be a noticeable performance hit.

I understand that this isn't an issue with your application, but any help will be greatly appreciated.

Thank you

foosel, 2017-03-10:
Yes, it does get called on every single line, and there will be a small performance hit, but it usually shouldn't be noticable. It could be limited further by matching against the connecting state of the printer maybe, to only look for the line if the printer is not yet in operational state if there is a recognizable performance hit.

sanchosk, 2018-02-04:
I had to use similar "nasty hack" for my Tronxy X5S that uses "dev" version of Marlin 1.1.7. Of course, the configs are not included, so I cannot rebuild it just like that.
However, changing the startup script in similar fashion to expect "echo: SD card init filed" fixed my connecting issue.
I'll recompile and upload a proper firmware into the board, but for now the dirty hack enabled me to print. Thanks!

superpomme, 2018-02-27:
Hi Sanchosk, I am also trying to get an x5s to connect to octopi, but having issues. I put the "echo: SD card init filed" in the startup script, but that didn't seem to work. I have the serial logs if that helps. Thanks, Rob

sanchosk, 2018-02-28:
There is a typo - not filed but failed :)
Try the new sentence and let me know.

Btw, I changed the electronics for RAMPS with TMC2208 and the printer is now quiet beyond believe and works with Octo no problem ;)

CarlosReyes07, 2018-04-09:
Hi @gayaths  do you have any problem after resolving the connection issue? I mean how is your machine performance?

Thanks in advance.

CarlosReyes07, 2018-04-17:
Hello everyone,
Somebody could help me to install this please?
Everytime i want to insert the plugin i have this message:
( sorry but i'm not a programmer and i have limited software knowledge)
"
Installing plugin from uploaded archive...
/home/pi/oprint/bin/python -m pip install /tmp/tmp2VeP2f.zip
Processing /tmp/tmp2VeP2f.zip
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-675Yph-build/
Complete output from command python setup.py egg_info:
Traceback (most recent call last):
File "<string>", line 1, in <module>
IOError: [Errno 2] No such file or directory: '/tmp/pip-675Yph-build/setup.py'
----------------------------------------
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
"