OctoPrint/OctoPrint #1817: Resend request - wrong (or no) line resent.

PJR-slo, 2017-03-10:
Firstly, apologies if I get this wrong; this is the first time I have filled in this type of report...  I am not an "expert" in computers and don't have a clue where to find things.  Sorry.

I have attempted to search for this same issue, but could not find anything current.

#### What were you doing?

Printing several GCode files from OctoPi/Print.  

#### What did you expect to happen?

I expected the model to print

#### What happened instead?

Communication issue (checksum mismatch) requesting a resend.  A resend does not appear to have happened.

#### Branch & Commit or Version of OctoPrint

Version 1.3.1 (Master Branch)

#### Printer model & used firmware incl. version

Prusa "original" i3 Mk2 with multi-material upgrade.  FW 3.0.10-Alpha (latest MM firmware)

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 56.0.2924.87 on Windows 7-64

#### Octoprint.log

just one example:

| Recv: ok
| Send: N8540 G1 X145.558 Y107.205 E0.09386 F1597.1*37
| Recv: ok
| Send: N8541 G1 X145.558 Y107.453 E0.08852 F1696.3*34
| Recv: ok
| Send: N8542 G1 X145.558 Y107.719 E0.08222 F1787.3*32
| Recv: ok
| Send: N8543 G1 X145.558 Y108.005 E0.07486 F1869*49
| Recv: ok
| Send: N8544 G1 X145.558 Y108.309 E0.06642 F1940.2*36
| Recv: ok
| Send: N8545 G1 X145.558 Y108.632 E0.05675 F1999.8*33
| Recv: Error:checksum mismatch, Last Line: 8541
| Recv: Resend: 8542
2017-03-09 10:01:34,438 - octoprint.util.comm - INFO - Communication timeout during an active resend, resending same line again to trigger response from printer.
2017-03-09 10:01:55,533 - tornado.access - WARNING - 404 GET /itassistant/ui/omaBaseFrame.htm (127.0.0.1) 20.18ms
2017-03-09 10:02:19,505 - octoprint.util.comm - INFO - Communication timeout during an active resend, resending same line again to trigger response from printer.
2017-03-09 10:03:00,039 - tornado.access - WARNING - 405 POST / (127.0.0.1) 18.67ms
2017-03-09 10:03:04,548 - octoprint.util.comm - INFO - Communication timeout during an active resend, resending same line again to trigger response from printer.
2017-03-09 10:03:04,809 - octoprint.filemanager.analysis - INFO - Starting analysis of local:PL1_PL2_PL3_PL4_MMPP.GCode

#### Contents of terminal tab

Recv: ok
Send: N8541 G1 X145.558 Y107.453 E0.08852 F1696.3*34
Recv: ok
Send: N8542 G1 X145.558 Y107.719 E0.08222 F1787.3*32
Recv: ok
Send: N8543 G1 X145.558 Y108.005 E0.07486 F1869*49
Recv: ok
Send: N8544 G1 X145.558 Y108.309 E0.06642 F1940.2*36
Recv: ok
Send: N8545 G1 X145.558 Y108.632 E0.05675 F1999.8*33
Recv: Error:checksum mismatch, Last Line: 8541
Recv: Resend: 8542
Recv: Error:No Line Number with checksum, Last Line: 8541
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N8541 G1 X145.558 Y107.453 E0.08852 F1696.3*34
Recv: Error:Line Number is not Last Line Number+1, Last Line: 8541
Recv: Resend: 8542
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N8541 G1 X145.558 Y107.453 E0.08852 F1696.3*34
Recv: Error:Line Number is not Last Line Number+1, Last Line: 8541
Recv: Resend: 8542

I have read the FAQ.

Peter

foosel, 2017-03-10:
Your printer requests a resend here, but never "clears" OctoPrint for sending, causing the timeout and the following issues - the `ok` after the resend request is missing. That's a bug in the printer's firmware.

OctoPrint has a setting to get around this firmware bug, try checking "Simulate an additional `ok` for resend requests" under "Settings" > "Serial Connection" > "Advanced options". Save and reconnect to your printer, see if that helps and please report back.

Additionally there indeed might be a mismatch though in the line sent by OctoPrint if it encounters a timeout during an active resend, I'll have to check the code for this very specific case.

PS: In the future, please provide the full `octoprint.log`, not just an excerpt from it. The log is being asked for because it contains valuable information about your specific setup (installed plugins, contextual information for errors and such). If you only provide an excerpt, chances are high you are stripping bits away that are crucial for being able to analyse your problem.

PJR-slo, 2017-03-10:
Many thanks for such a quick response!  I have forwarded this issue to the PR gurus.

Apologies for not providing complete info; I am pretty certain there will not be a "next time", but if there is I will try harder!.

Peter

closed 2017-03-10 by PJR-slo

thinkyhead, 2018-03-11:
> never "clears" OctoPrint for sending

Is Marlin supposed to send this in response to an error…
```
Line Number is not Last Line Number+1, Last Line: ###
Resend: ###
ok
```
…or some other variant?

And the same in response to `M999`?