OctoPrint/OctoPrint #1818: Relative E-distances not correctly displayed in gcode preview

VBen, 2017-03-11:
#### What were you doing?
I'm using a printer with two seperate nozzles. The nozzles are magnatical dockabel.
Due to my preprocessing i have to use relative E distances.

#### What did you expect to happen?
The gcode display should work in the same was as with absolute e distances

#### What happened instead?
The relative E-distances are not correctly displayed in the gcode viewer.
The printing works just fine.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.2rc1 (rc/maintenance branch)

#### Printer model & used firmware incl. version

Homebuild , running Marlin 1.1.0-rc8 on a RAMPS board

#### Browser and Version of Browser, Operating System running Browser
Does not matter, I get the same result in every browser I tested

#### Link to octoprint.log

https://gist.github.com/VBen/e4fbe3f02b3469b0e892464e3de4a7da

#### Link to contents of terminal tab or serial.log
no communication errors

#### Link to contents of Javascript console in the browser

I think its not necessary

#### Screenshot(s) showing the problem:

[image]

#### Used Gcode File
 See [https://gist.github.com/VBen/e4fbe3f02b3469b0e892464e3de4a7da](https://gist.github.com/VBen/e4fbe3f02b3469b0e892464e3de4a7da)

VBen, 2017-03-12:
I did some further investigation
It seems that this behaviour is Marlin related. Marlin seems to keep the status(M83) of relative E ditances  over a G91/G90 mode switch.
But the gcode parser always resets the E distances to absolute on every G90 command
[https://github.com/foosel/OctoPrint/blob/master/src/octoprint/static/gcodeviewer/js/Worker.js#L366](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/static/gcodeviewer/js/Worker.js#L366)

christianh17, 2017-03-27:
Suggestion: Make a switch so everyone can decide which way octoprint should "think".

foosel, 2017-04-10:
The [RepRap wiki's GCODE node](http://reprap.org/wiki/Gcode#G90:_Set_to_Absolute_Positioning) (which is the closest thing available to a protocol definition - sadly), states this about `G90`:

> All coordinates from now on are absolute relative to the origin of the machine.

That _sounds_ like it also includes the `E` axis and the current implementation would therefore be correct (and Marlin's interpretation be wrong). 

Of course, we don't have a proper standard (hooray...), so all is just guess work. And since there's absolutely no way to determine what is the correct way dynamically, I fear there really is no way around yet another one of those horrible checkboxes that the settings dialog is already drowning in.

foosel, 2017-05-10:
[image]

After some reading in various firmware sources (Marlin old and new, Repetier, Smoothieware) I've now changed the behaviour to mirror Marlin and Repetier by default (= `G90` doesn't disable relative extrusion mode previously enabled through `M83`), with an option to switch to Smoothieware style behavior (= `G90` overwrites relative extrusion mode, so a new `M83` is needed after each `G90`) under Settings > Features. 

Available on the `maintenance` branch, soon also `devel` and to be released with 1.3.3.

I have to admit that I'm not too happy about having to add yet another checkbox for that, but considering that we have two approaches in popular firmware here that are exactly the opposite of each other, there's not much choice.

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel