OctoPrint/OctoPrint #1819: timelapse for disconnected printer ?

goffs, 2017-03-12:
is there any possibility to use the timelapse functionality 
of octopi/octoprint in combination with the davinci pro printer ?

we would like to timelapse even when octoprint is 
unable to connect to the printer ... as this particular
printer-type does not support host-mode in stock fw.

see also
https://github.com/luc-github/Repetier-Firmware-0.92/issues/151
for a reason, why this is so.

ntoff, 2017-03-12:
what are you using octoprint for then? Just asking because there are standalone ways to do timelapses so if you don't need octoprint at all, you can use a bash script on linux, various webcam softwares under windows / osx.

Anyway, you can set up a virtual printer in octoprint, connect to that, then run your gcode and turn timelapse on. Be aware that octoprint won't know whether your printer is still running if it's not actually connected to it so if octoprint finishes its fake print before your printer is done (I assume you're printing from an sd card or something?), the timelapse will end prematurely.

See here for how to enable the virtual printer: http://docs.octoprint.org/en/master/configuration/config_yaml.html#development-settings

and assuming you're using a raspberry pi, see here for fswebcam for taking shots at a timed interval https://www.raspberrypi.org/documentation/usage/webcams/ which you'd then convert into a video either with avconv on the pi or in a video editing software suite like adobe premiere or sony vegas if you need fancy titles and overlays, or even  something simple like virtualdub if you just want a simple video.

Or you can use raspistill if you're using a raspberry pi camera module.

goffs, 2017-03-12:
cool ... thanks for your very informative answer :-)

closed 2017-03-12 by goffs