OctoPrint/OctoPrint #1820: restarted and octoprint would not start see bottom of page

nowkillkennys, 2017-03-13:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log
at the bottom of page
[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:
octoprint wouldn't start

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

2017-03-12 23:35:48,278 - octoprint.server - INFO - Starting OctoPrint 1.3.1 (master branch) 2017-03-12 23:35:48,367 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages... 2017-03-12 23:35:51,249 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 16 hook handlers 2017-03-12 23:35:51,284 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages... 2017-03-12 23:35:51,377 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 16 hook handlers 2017-03-12 23:35:51,384 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads... 2017-03-12 23:35:51,385 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully. 2017-03-12 23:35:51,391 - octoprint.plugins.nautilus - INFO - Nautilus - OctoPrint mobile shell, started. 2017-03-12 23:35:53,491 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip 2017-03-12 23:35:55,582 - octoprint.util.pip - INFO - Version of pip is 9.0.1 2017-03-12 23:35:55,583 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes 2017-03-12 23:35:55,594 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s) 2017-03-12 23:35:55,597 - octoprint.plugins.nautilus - INFO - Migrate settings from None to 6. 2017-03-12 23:35:55,623 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system: | Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/announcements | Autoselect Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoselect | Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/corewizard | Cost Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cost | CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/cura | Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/discovery | EEPROM Marlin Editor Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin | FileManager (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filemanager | Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar | Nautilus (1.10) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_nautilus | Navbar Temperature Plugin (0.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp | Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/pluginmanager | Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/softwareupdate | System Command Editor (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_systemcommandeditor | TouchUI (0.3.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui | Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins/virtual_printer | Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher 2017-03-12 23:35:55,625 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue... 2017-03-12 23:35:55,632 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue 2017-03-12 23:35:55,633 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets... 2017-03-12 23:35:55,634 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache... 2017-03-12 23:35:56,317 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp 2017-03-12 23:35:56,330 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp 2017-03-12 23:35:56,334 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP 2017-03-12 23:35:56,407 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json 2017-03-12 23:35:56,453 - octoprint.server - INFO - Listening on http://0.0.0.0:5000 2017-03-12 23:35:56,664 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.2s 2017-03-12 23:35:57,792 - octoprint.plugins.nautilus - INFO - Logging level is INFO. 2017-03-12 23:36:09,118 - octoprint.server - INFO - Shutting down... 2017-03-12 23:36:09,340 - octoprint.events - INFO - Processing shutdown event, this will be our last event 2017-03-12 23:36:09,342 - octoprint.events - INFO - Event loop shut down 2017-03-12 23:36:09,359 - octoprint.server - INFO - Goodbye! 2017-03-12 23:36:13,893 - octoprint.server - INFO - Starting OctoPrint 1.3.1 (master branch) 2017-03-12 23:36:13,962 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins, /home/pi/.octopr [... cut ...] 2017-03-12 23:58:00,705 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages... 2017-03-12 23:58:00,805 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 16 hook handlers 2017-03-12 23:58:00,962 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json

ntoff, 2017-03-13:
@nowkillkennys Please fill out the ticket properly, that includes ALL the information you had to scroll passed to paste your log. We need to know what you were doing before the error occurred, were you editing files? Did you perform an update in some way? All that information is necessary.

nowkillkennys, 2017-03-13:
I added the YouTube-upload the one on the cookbook wiki and installed file manager also naultils came up with an update I updated looked like it updated something els aswell and then restarted and well then not much happend

nowkillkennys, 2017-03-13:
I got my mate to get it back isn't really the same he moved some of it to a different area.
 I've just looked in the update/plugins manger area and I've just seen octoprint 1.3.1! INSTALLED:UNKNOWN  AVAILABLE:UNKNOWN

nowkillkennys, 2017-03-13:
I used this as-well would of this of been a cause ?
Automatically upload timelapses to YouTube
Using youtube-upload and the MovieDone event hook it is possible to automatically upload finished timelapses to YouTube.
You'll need to install youtube-upload first. Assuming you are on a Pi, this can accomplished via
sudo apt-get install python-gdata
wget https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/youtube-upload/youtube-upload-0.7.3.tgz   
tar xfz youtube-upload-0.7.3.tgz
cd youtube-upload-0.7.3
sudo python setup.py install
After that, you'll need to add the following event subscription to events > subscriptions in your config.yaml
- command: 'youtube-upload --email="<yourUser>" --password="<yourPassword>" 
    --title="Timelapse: {gcode}" --description="Timelapse of {gcode}, printed and 
    recorded via OctoPrint" --category="Tech" --keywords=OctoPrint "{movie}"'
  event: MovieDone
  type: system
Substitute <yourUser> and <yourPassword> with your YouTube username and password respectively. If you are using Google's two factor authentication, you'll need to create an application password to use here. If you want to make your movie unlisted, add --unlisted after the keywords, if you want to make it private, add --private.

youtube-upload-0.7.3.tgz

ntoff, 2017-03-13:
>I got my mate to get it back isn't really the same he moved some of it to a different area.

You shouldn't need to  move anything to any other area, octoprint relies on the directory structure staying the way it is. At this point it's probably easier to wipe your SD card and start fresh.

Remember that when editing the config.yaml file, even just one out of place tab stop can prevent octoprint from loading, and OSX can cause issues with weird quote marks, i.e. the " marks aren't standard quotation marks, it's best to SSH into the pi and edit the config.yaml file through "nano" (an editor preinstalled), the config.yaml patcher plugin through octoprint, or an editor under OSX that doesn't use funky quotation marks, that's if you are using OSX (just throwing that out there).

Wipe your SD card, reflash the octopi image, then add back your plugins one by one, restart between each plugin install until you find one that breaks something. Then, if something does go wrong, document the process you used exactly and fill out the ticket template properly, filling in all the OTHER things you've ignored (twice now), and add to that list of stuff, which editor you're using to edit config.yaml (as some editors can mangle it).

Just to add one more thing:
>sudo python setup.py install

Assuming that's to install an octoprint plugin, you shouldn't be running sudo before anything to do with octoprint if you're using the octopi prebuilt image, it runs in a virtual environment, you need to run /home/pi/oprint/bin/python setup.py install after cd'ing into the plugin's directory.

foosel, 2017-03-13:
What @ntoff said. It's impossible to even figure out what problem you are running into based on the information you provided. I understand you are new to this, but there is a lot of information provided on where to find all the data that you need to provide in a ticket - you scrolled right past it and apparently also didn't click the big fat yellow box on the top of the ticket template that said information you scrolled past also repeatedly hinted you at to get help with how to open a proper ticket. It's ok to not knowing how stuff works, but please read information provided to you that can help you with this initial hurdle.

I'm closing this now, feel free to request a reopen if your problem persists after what @ntoff suggested and you have provided a *fully filled out ticket template*.

This is not to annoy you, it's simply impossible to help you without the required information.

closed 2017-03-13 by foosel

nowkillkennys, 2017-03-13:
Right ok I do apologise sincerely I will get my brother-in law  to grab the files from the erroerd files and will do a new post/issue I'll try to include the yaml. octoprint.log and the 3 things that happens before it went pear-shaped