OctoPrint/OctoPrint #1823: rendering message

overpaul, 2017-03-16:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete any lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to repor tis still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the contribution
guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please describe the steps to reproduce your issue. Be as specific as
possible here. The maintainers will need to reproduce your issue in
order to fix it and that is not possible if they don't know what you
did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen and what happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure or reporting UI issues.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting any kind of communication issues between OctoPrint and your printer.
Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

overpaul, 2017-03-16:
Very minor suggestion, the message that is given when converting the stills to MPEG is misleading.
It talks about post processing which I assumed was the post roll capture.  it would be better just to say it will take till.... TO RENDER the video.

closed 2017-03-16 by overpaul

foosel, 2017-03-16:
I'm sorry, but I don't know what you are referring to. The notification about it capturing the post roll images is "Now capturing timelapse post roll, this will take approximately xyz s...", the one about rendering the movie is "Now rendering timelapse foo.mpg. Due to performance reasons it is not recommended to start a print job while a movie is still rendering.", nothing about "post processing".

Third party plugin? Maybe a screenshot would help.

overpaul, 2017-03-17:
I didn't have the exact message in front of me when I wrote you. I had time
lapse post roll set at 1 second and the message show almost 30 minutes to
do that,  I don't believe I ever saw the second message.  I have played
around with the time lapse post roll time to reduce the time shown in the
capturing timelapse post roll message but it always shows a long time.

Perhaps if you ask for a post roll you only get the first message which
combines the post roll time and the rendering time.

When I had the post roll message I looked at the temp folder on the Pi and
saw that the number of files decreasing, so I suspect it was rendering.

Upgraded to the latest version and will watch the messages after my print
job completes.  I will send screen shots.

Great program, by the way.  It really changed the way I printed and allow
me to move the printer to a more ventilated area.

Thanks for all you are doing.

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

overpaul, 2017-03-17:
Performed as you said it would.  See screen shots.
My only concern is why it would take 4 minutes to prepare the post roll of
1 second.
I had the post roll time set to 30 seconds before and the time to generate
it was quite long.  Probably never saw the second message on rendering.

Sorry for the confusion on my part

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600

overpaul, 2017-03-17:
forgot the screen shots

*Paul Oversmith*

E-Mail: pgo@overweb.com
Phone: (941) 266-3600