OctoPrint/OctoPrint #1827: Plugins not installing jessie 0.13

zakth, 2017-03-20:
Greetings, I've recently updated Octopi to 0.13 (jessie) and I cannot get any plugins to install. Below is the full error:

Exception:
Traceback (most recent call last):
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 209, in main
status = self.run(options, args)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 299, in run
requirement_set.prepare_files(finder)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 360, in prepare_files
ignore_dependencies=self.ignore_dependencies))
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 577, in _prepare_file
session=self.session, hashes=hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 810, in unpack_url
hashes=hashes
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 649, in unpack_http_url
hashes)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 842, in _download_http_url
stream=True,
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 480, in get
return self.request('GET', url, **kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 378, in request
return super(PipSession, self).request(method, url, *args, **kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 468, in request
resp = self.send(prep, **send_kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 597, in send
history = [resp for resp in gen] if allow_redirects else []
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 195, in resolve_redirects
**adapter_kwargs
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 576, in send
r = adapter.send(request, **kwargs)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 46, in send
resp = super(CacheControlAdapter, self).send(request, **kw)
File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 447, in send
raise SSLError(e, request=request)
SSLError: ("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",)

therealjkvalentine, 2017-03-24:
Having same issue, will attempt to complete the ticket.

#### What were you doing?

Attempting to install plugins (e.g. slack or email notifier) from GUI and command line via ssh

#### What did you expect to happen?

Plugin success message and ability to use plugin after installation

#### What happened instead?

Error message, appearing to indicate that the SSL certificate could not be verified (similar to this issue with another project, solved by changing certificate authority bundle: https://github.com/mesosphere/marathon-lb/issues/178)

#### Branch & Commit or Version of OctoPrint

Version: 1.2.10 (master branch)

#### Link to octoprint.log

[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/866601/octoprint.log.txt)

#### Screenshot(s) showing the problem:

[image]
[image]
[image]

I have read the FAQ.

Thank you for your time and dedication to this wonderful project! Proud to be a supporter! 

Best,
Jim

foosel, 2017-03-24:
> 2016-03-18 08:17:16,602 - octoprint.server - INFO - Starting OctoPrint 1.2.10 (master branch)

The operating system beneath OctoPrint (OctoPi?) thinks it's 2016. SSL requires the system date to be correct, otherwise certificates cannot be validated properly - that is what's happening here.

I'm assuming you are running OctoPi here: Usually with a working internet connection it should update its system date via ntp pretty much directly on first boot. For some reason that didn't happen here. If this is a Pi3 I'd guess [it's this issue](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141454), in which case adding this line just before the `exit 0` in `/etc/rc.local` should solve it:

    /sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00

This can also be done in one command from via SSH:

    sudo sed -i 's@exit 0@/sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00\nexit 0' /etc/rc.local

**Note** This is one line! No line breaks!

In any case, not an issue with OctoPrint but with a wonky system date, probably due to a bug in the RPi3 wifi driver. I've tested against an OctoPi 0.14 image I built manually a while ago and with that I cannot reproduce the issue so I guess the driver shipped with current Raspbian does no longer filter NTP requests, even though the [corresponding ticket](https://github.com/raspberrypi/linux/issues/1519) is still open, so to be extra sure I'll file a PR against that to pull the above line into the next stable image.

closed 2017-03-24 by foosel

therealjkvalentine, 2017-03-24:
Glorious, thank you so much. Sorry for misfiling as bug, hopefully others will come across this when searching for the solution :)

PS- Yes, this is an octopi build and yes I have been trying to figure out the system date issue this morning not realizing they were (obviously in hindsight ;) related.

foosel, 2017-03-24:
Let's hope for the best, past experience sadly has shown that people rarely search/google first, but hope dies last 😄