OctoPrint/OctoPrint #1828: [Request] Visualize Firmware Retract/Restore

tcm0116, 2017-03-22:
#### What were you doing?

When visualizing a gcode file generated using G10/G11 retractions, the retractions are not shown in OctoPrint's GCode Viewer.

#### What did you expect to happen and what happened instead?

I expected to see the retract/restore visualizations, but none were displayed.

#### Branch & Commit or Version of OctoPrint

1.3.1

#### Operating System running OctoPrint

Raspain

#### Printer model & used firmware incl. version

HICTOP 3DP-11 (modified)

#### Browser and Version of Browser, Operating System running Browser

Safari 10.0.3 on macOS 10.12.3

ParkerK, 2017-04-11:
Can you share a screenshot perhaps, and the associated GCode?

tcm0116, 2017-04-11:
A screenshot would be pretty boring since you would just see lines with no retracts. The GCode can be generated by selecting the "Use Firmware Retraction" option in Slic3r, which causes it to insert G10/G11 in the GCode instead of G0/G1 commands with just an E parameter.

Depaulicious, 2017-05-17:
The retractions are shown in the GCode viewer. As @tcm0116 said, make sure you enable the option in your slicer, and that the "show retraction" check mark in the GCode viewer is checked.

easytarget, 2018-02-15:
With 'show retractions' enabled in the viewer, and a file sliced in slic3r with the 'firmware retract' check-box enabled (eg uses G10/11 rather then extruder/zmoves) you do not see these retracts. 

Exactly the same model sliced with same settings apart from that checkbox disabled shows all retracts correctly.

Firmware retraction now supported by all major slicers and firmwares, G10 causes a retract, G11 resumes. And it's becoming more popular as people catch on to it's advantages (faster, on the fly tunable, portable)

easytarget, 2018-02-15:
Doesn't look to difficult to add to gcodeviewer/js/Worker.js, I'm a lousy programmer but adding the processing/tests on G10/11 was simple, I need to test locally (and run the unit tests) once I'm home and the printers are idle. If it works I'll generate a PR

foosel, 2018-02-15:
@easytarget thank you, that sounds great!

AnSch1510, 2022-01-05:
any progress on this so far?

easytarget, 2022-01-10:
Sorry; I did spend an evening on it years go, didnt know Python so well, and it didn't work (despite my code looking correct to me, the retract/advance moves were not shown). I gave up on it, apologies for not updating here. IIRC it did not look too hard to do.