OctoPrint/OctoPrint #1829: [Request] USB light source control

thhart, 2017-03-22:
I use a simple USB led lamp for illuminating the printer and observing printing remotely for long print jobs. It would be nice to be able to control this light via the interface. This means the USB bus might have been de-/activated in the underlying system (Raspberry in my case). Further more the light might be triggered as a short flash for timed image records, but this is only a next request.

foosel, 2017-03-22:
That is too hardware specific for something that is supposed to be platform agnostic (which OctoPrint targets to be). 

It sounds like something that - if you find a way to toggle power to a particular USB socket without interfering with the other ports, which [appears to be quite the challenge](https://superuser.com/questions/314402/disable-usb-powered-light-from-command-line) - you could already do via [custom system commands](http://docs.octoprint.org/en/master/configuration/config_yaml.html#system) (easier accessible via [the System Command Editor plugin](http://plugins.octoprint.org/plugins/systemcommandeditor/)).

tldr: The challenge here is something outside of OctoPrint, and wiring up a light toggle in OctoPrint once you've found a solution for that challenge is already possible.

closed 2017-03-22 by foosel

nophead, 2017-03-22:
I don't know if it is even possible on an RPI. The original B has the VBUS of the USB ports hard wired to 5V. They don't release full schematics for the later ones.