OctoPrint/OctoPrint #1831: OctoPrint loses connection to printer when printing with variable layer hight

sch00f, 2017-03-23:
#### What were you doing?

I've been using OctoPrinnt for a couple of weeks now, no problems at all. It works like a charm, for 5 minute prints just as well as with prints that take 10+ hours.
But when I tried the new variable layer hight feature in Slic3r OctoPrint somehow couldn't handle it.

#### What did you expect to happen and what happened instead?

Expected the print to start like normal, but instead OctoPrint freezes right after hitting the print button.
Temperature doesn't get updated, printer doesn't heat up or move, but the webcam still works.
After a minute or so a connection timeout error happens, and after reconnecting to the printer and aborting the job it communicates again.
Other files still work as expected.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint

OctoPi on a Raspberry Pi 3

#### Printer model & used firmware incl. version

Prusa i3 mk2

#### Link to octoprint.log

https://gist.github.com/sch00f/f972ddf26f30ca3bedce16cb15a20694

#### Link to contents of terminal tab

https://gist.github.com/sch00f/2435501f5f605ba0f6e193d5c1e5541b

I have read the FAQ.

nophead, 2017-03-23:
Your printer never replies after getting N2 M115 U3.0.10*83
M115 is get firmware capabilities and doesn't take any parameters as far as I know.

ntoff, 2017-03-24:
@nophead That's for the genuine prusa mk2 firmware

https://github.com/prusa3d/Prusa-Firmware/blob/12505d0f4ed588b66571ecac94b17c4e2f35a6f2/Firmware/Marlin_main.cpp#L4167

nophead, 2017-03-24:
Looks like the printer might be waiting for a firmware upgrade then.

foosel, 2017-03-24:
Agreed. Firmware is probably blocking with a message on the LCD. Nothing OctoPrint can do anything about, if it stops responding on serial it's basically dead to any host depending on the serial interface.

Also overloading `M115` like this is a very bad idea. I opened a ticket on the Prusa firmware repo (see above).

I'm closing this ticket here though because the behaviour in question is not caused by OctoPrint but rather by non-standard printer behaviour, in this particular problem probably solved by updating the printer's firmware to the version required by the sliced GCODE.

closed 2017-03-24 by foosel