OctoPrint/OctoPrint #1832: [request] automatically collapse the files accordion when file loaded and print started?

ntoff, 2017-03-24:
request / idea

Just like when a connection to the printer is made, the connection accordion collapses itself, I think it would be nice if the "Files" section were to auto collapse once you've loaded a file and started a print. 

I don't think it should depend just on loading a file since you could easily load the wrong one and have it collapse, then have to open it again to get back at it so having it auto collapse, so once a print starts seems like a better idea since you're probably done with that section. Then, manually re-opening it would leave it open (no auto re-close timer).

foosel, 2017-03-28:
This should be doable through a plugin and actually is something I'd say should only be done through a plugin. The reason being that while the connection pane has absolutely no use at all once you are connected besides for disconnecting again (which is a disruptive change enough that the additional click is a good idea there), the file list still might be needed during a print for preparing the next one, looking up something and so on. I know I use it like that occasionally, and I'd imagine it becoming annoying quite quickly if collapsing was the default behaviour. And I've also become a bit careful about adding more and more checkboxes for very specific behaviour, I don't want things to end up even more looking like [this](https://pbs.twimg.com/media/BylctRxCEAEv-i2.png) ;)

closed 2017-05-07 by ntoff

ntoff, 2017-11-28:
A plugin to do this is now available at https://github.com/ntoff/OctoPrint-AutoCollapse

Auto collapse timeout is user configurable (defaults to 30 seconds) and hovering over the accordion will disable the timer and prevent auto  hiding in case the user happens to be interacting with it. Leaving focus will *not* re-enable the timer though, I did experiment with that but none of the other accordions auto hide so it was a bit weird having the files accordion do it.