OctoPrint/OctoPrint #1833: Issue with Custom labelled ports (Linux) with Smoothieware

AxMod3DPrint, 2017-03-24:
#### What were you doing?
Printer disconnected from Octoprint due to a time-out after a M81. Octoprint tries to reconnect but only half connects. It does connect and read from the USB on the terminal (M105, SD File list) but no controls are active. Restarting the Octoprint instance usually allows for a clean connect, Generally happens when printer is idle or after a M112 

#### What did you expect to happen and what happened instead?
Connect as normal

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.2 (master branch)

#### Operating System running OctoPrint
Ubuntu Server 16.04.2 LTS (GNU/Linux 4.6.0-040600-generic x86_64) - multiple OP instances run on this box but only the 2 Smoothieware, printers are affected, the Marlin Based Printers are fine. Custom named persistent USB-Serial port names are used to avoid confusion when connecting to the printer.

#### Printer model & used firmware incl. version
FB2020 CoreXY/AX-02 Delta. Smoothieware any current version happens with current release version to current edge version

#### Browser and Version of Browser, Operating System running Browser
Linux Mint/Opera 44 Windows/Opera 44 Bohdi Linux/Firefox 45

#### Link to octoprint.log
http://pastebin.com/MV3axDTk

#### Link to contents of terminal tab or serial.log
http://pastebin.com/pYjh3ewH - from terminal but OP thinks not connected but is reading temps via M105

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) or video(s) showing the problem:
https://youtu.be/V3HBpYuf1WU 

I have read the FAQ.

foosel, 2017-03-24:
Hm. That sounds like something in the internal state got horribly confused - it still has a connection open but thinks it doesn't, hence it is offering you to connect. I have no idea right now how that would even be possible so we have to dig deeper.

You say it got into this state after a timeout following an `M81`. Can you reproduce this behaviour and provide the log output on the terminal from the exact moment that issue happens? The terminal output you provided sadly doesn't include the cause of the problem, only the symptoms and is not much use here.

Would you also be able to point out at which point in your provided `octoprint.log` the problem started/the timeout happened? Because that doesn't show any timeout at all.

A video of the reproduction itself might also be very helpful here - the one you posted sadly again shows the symptoms after the problem started basically, not the actual moment where stuff went south.

ntoff, 2017-03-24:
I had a similar thing when I gave my printer an M112 during a blocking heatup cycle, didn't do anything much (not octoprint's fault) but that M112 caused octoprint to automatically disconnect and upon trying to reconnect it only half succeeded. I got the temperature responses the printer was automatically sending out (repetier firmware, it was still trying to  heat up) but octoprint's connection status was sort of left in limbo. Rather than restart octoprint, a reset of my control board fixed it. Sorry, I didn't bother saving any logs as I figured it was more a firmware issue or an issue with my board (usually a connection attempt will reset it, for some reason this  time it didn't)

AxMod3DPrint, 2017-03-24:
@foosel Give me a while, I've not actually paid much attention to actually how to reproduce it, so I'm not exactly sure how to. Let me switch the serial logging on and see if I can find you something as I've checked a few times and nothing seems to appear in the main log. It can be from an idle printer or from a pulled USB cable, M81 or from as @ntoff said above an M112. It's mainly when the USB-Serial link is severed abruptly.

It seems a little random, but from experience, it's most probably not, as to how and when it does, having switched one of the Smoothie printers back to the standard /dev/ACM* it's fine.

AxMod3DPrint, 2017-03-24:
Ok, after a bit of playing and trying to replicate the issue, I've not been able to totally replicate the issue, but please keep this on hold for the moment. I've swapped out the cable on one of the printers to eliminate that and during that operation I had the following happen:
~~~
Connecting to: /dev/AX02
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x7fabac4f3890, open=True>(port='/dev/AX02', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: k
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x7fabac4f9990, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: o
~~~

It didn't want to connect at and was only partially receiving the first OK before disconnecting. This happened with both the old and new cables. I have also tried to eliminate any issues with the port naming by removing the tty- from the beginning, but as you can see from the above log a similar issue occurred. At this point I'm thinking it's more the hardware it's running on, an old HTPC which can get a bit on the warm side causing communication issues, but I may be wrong. Bare with me and I'll see if I can pin this down a little more.

AxMod3DPrint, 2017-03-25:
Right, replicated.  Had pulled the USB Cable to power down the printer and pull the SD card to work and check on the config. From terminal:
~~~
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792
Changing monitoring state from 'Connecting' to 'Operational'
Changing monitoring state from 'Operational' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1792'
Send: N0 M110 N0*125
Connection closed, closing down monitor
Recv: ok
Send: N1 M115*39
Recv: ok
Send: N2 M21*18
Recv: SD card ok
Recv: ok
Send: N3 M117 Hathor Connected*80
Recv: ok
Send: N4 M20*21
Recv: Begin file list
Recv: config
Recv: End file list
Recv: ok
Send: N5 M105*34
Recv: ok T0:18.5 /0.0 @0 B:18.0 /0.0 @0
~~~

At this point it did what is shown in the video

The Octoprint Log shows the following

~~~
Traceback (most recent call last):
  File "/home/ax/hathor/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1792, in _readline
    ret = self._serial.readline()
  File "build/bdist.linux-x86_64/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2017-03-25 21:30:44,855 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/Hathor', 'autoconnect': True, 'baudrates': [115200, 250000, 230400, 57600, 38400, 19200, 9600], 'ports': ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0'], 'baudratePreference': 115200}
2017-03-25 21:30:44,855 - octoprint.plugins.portlister - INFO - Checking if /dev/ttyACM1 is in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:44,857 - octoprint.plugins.portlister - INFO - autoconnect_delay 20
2017-03-25 21:30:44,865 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/Hathor', 'autoconnect': True, 'baudrates': [115200, 250000, 230400, 57600, 38400, 19200, 9600], 'ports': ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0'], 'baudratePreference': 115200}
2017-03-25 21:30:44,866 - octoprint.plugins.portlister - INFO - Checking if /dev/Hathor is in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:44,868 - octoprint.plugins.portlister - INFO - autoconnect_delay 20
2017-03-25 21:30:45,889 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/Hathor', 'autoconnect': True, 'baudrates': [115200, 250000, 230400, 57600, 38400, 19200, 9600], 'ports': ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0'], 'baudratePreference': 115200}
2017-03-25 21:30:45,892 - octoprint.plugins.portlister - INFO - Checking if /dev/sg1 is in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:45,893 - octoprint.plugins.portlister - WARNING - Won't autoconnect because /dev/sg1 isn't in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:45,897 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/Hathor', 'autoconnect': True, 'baudrates': [115200, 250000, 230400, 57600, 38400, 19200, 9600], 'ports': ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0'], 'baudratePreference': 115200}
2017-03-25 21:30:45,898 - octoprint.plugins.portlister - INFO - Checking if /dev/sdb is in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:45,898 - octoprint.plugins.portlister - WARNING - Won't autoconnect because /dev/sdb isn't in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:45,934 - octoprint.plugins.portlister - INFO - on_port_created connection_options {'portPreference': '/dev/Hathor', 'autoconnect': True, 'baudrates': [115200, 250000, 230400, 57600, 38400, 19200, 9600], 'ports': ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0'], 'baudratePreference': 115200}
2017-03-25 21:30:45,935 - octoprint.plugins.portlister - INFO - Checking if /dev/sdb1 is in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:30:45,936 - octoprint.plugins.portlister - WARNING - Won't autoconnect because /dev/sdb1 isn't in ['/dev/Hathor', '/dev/ttyUSB0', '/dev/ttyACM1', '/dev/ttyACM2', '/dev/ttyACM0']
2017-03-25 21:31:04,862 - octoprint.plugins.portlister - INFO - do_auto_connect
2017-03-25 21:31:04,863 - octoprint.plugins.portlister - INFO - realpath match
2017-03-25 21:31:04,866 - octoprint.plugins.portlister - INFO - Attempting to connect to /dev/Hathor at 115200 with profile '_default'
2017-03-25 21:31:04,870 - octoprint.plugins.portlister - INFO - do_auto_connect
2017-03-25 21:31:04,871 - octoprint.plugins.portlister - INFO - realpath match
2017-03-25 21:31:04,874 - octoprint.plugins.portlister - INFO - Attempting to connect to /dev/Hathor at 115200 with profile '_default'
[... cut ...]
  File "build/bdist.linux-x86_64/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
~~~

I had clicked to connect, I may be wrong but it's looking like the Portlister Plugin is at fault here trying to connect at the same time and causing the time-out. I've disabled it on one instance to check if it is the issue.

AxMod3DPrint, 2017-03-26:
@foosel - after further testing with the Port Lister plugin disabled, it seems to have sorted it. So it seems to be an issue with the Port Lister plugin, and not Octoprint itself.

foosel, 2017-03-27:
@markwal any ideas what in the plugin could be causing something like that? O_O I thought it was only refreshing the list of serial ports?

markwal, 2017-03-27:
The PortLister also has an autoconnect feature that can be turned off. It looks like there's a race here. It doesn't attempt autoconnect if a connection is already made, but if two printers show up simultaneously that check may fail since it probably looks for connected instead of !Offline.

Moving to PortLister tracker.

closed 2017-03-27 by markwal